wasm-micro-runtime/wasm-micro-runtime #1348: LLVM ERROR: IMAGE_REL_AMD64_ADDR32NB relocation requires an ordered section layout

abc767234318, 2022-08-01:
I execute the following files in JIT mode.
[demo.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/9234339/demo.zip)
It prints the following error.
```
Exception: lookup the entry point symbol (like _start, main, _main, __main_argc_argv) failed
LLVM ERROR: IMAGE_REL_AMD64_ADDR32NB relocation requires an ordered section layout
```
However, no LLVM errors are printed in wasmtime and wasmer. I wonder if this is WAMR's problem.

wenyongh, 2022-08-04:
Hi, we have a test, but cannot reproduce the issue, wondering which JIT mode is used, and which platform, which arch?

abc767234318, 2022-08-08:
JIT mode: LLVM JIT
Platform: Windows 11
Arch: x86
@wenyongh

wenyongh, 2022-08-10:
@abc767234318 Thanks for reporting the issue, I submitted PR #1372 to fix it, please try again.

wenyongh, 2022-09-22:
Close this issue as it was fixed.

closed 2022-09-22 by wenyongh