wasm-micro-runtime/wasm-micro-runtime #1349: Exception: wasm operand stack overflow ?

Zhangyx24, 2022-08-02:
## Steps to reproduce
 Trying to execute a QuickJS engine with WAMR.
[qjs.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9238471/qjs.txt)
The qjs.wasm is uploaded as qjs.txt.

three mode:
（1）interpreter : `iwasm qjs.wasm`
（2）JIT: `cd JITbuild ` and `iwasm qjs.wasm`
（3）AOT: `wamrc -o qjs-wamr.aot qjs.wasm` and `iwasm qjs-wamr.aot`

## Expected result
Successfully execute the qjs.wasm in all the three mode and print:
"QuickJS - Type "\h" for help
qjs > ^C"

## Actual result
（2）and （3）could successfully execute, but （1）got some problem with the output:
Exception: wasm operand stack overflow

wenyongh, 2022-08-04:
Hi, wasm operand stack overflow was thrown due to not enough stack size, please use `iwasm --stack-size=n` to enlarge it, for example:
```
iwasm --stack-size=102400 qjs.wasm
```

Zhangyx24, 2022-08-06:
Thanks a lot !

closed 2022-08-06 by Zhangyx24