wasm-micro-runtime/wasm-micro-runtime #1357: Homebrew formula

bonifaido, 2022-08-05:
I'm in the progress of submitting a PR with a WAMR Homebrew formula here: https://github.com/Homebrew/homebrew-core/pull/107301

As you can see in the review, we are having certain issues there, but the main blocker is that the project doesn't have semver, but rather a calver versioning scheme, which itself is not an issue, but the current calver format is not lexicographically correct, which matters for Homebrew. Can we change this versioning scheme to YYYY.MM.DD format. Or maybe you have some plans for this already, which might be better?

Also please if you have any insights other than this connected to the formula feel free to comment!

lum1n0us, 2022-08-07:
it is a nice one and we will keep you updated.

lum1n0us, 2022-08-08:
Btw, is it about the latest release [fast-jit-06-29-2022](https://github.com/bytecodealliance/wasm-micro-runtime/releases/tag/fast-jit-06-29-2022) not using the usual "WAMR-MM-DD-YYYY" pattern?

bonifaido, 2022-08-08:
Yes, that is also an issue, but the original issue is that it would be better to use the YYYY-MM-DD CalVer pattern, so tools like Homebrew could detect version changes based on lexicographical order, for example according to the current format 01-21-2021 is newer than 05-13-2022 which is an issue for homebrew.

carlocab, 2022-08-11:
To be clear: we're not asking you to change the version scheme for your project.

We were just looking for feedback about whether using your existing version scheme to construct a CalVer scheme is a sensible move for us. It wouldn't be if, for example, you intend to adopt a different version scheme in the near future.

lum1n0us, 2022-08-12:
We are changing:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/1374

bonifaido, 2022-08-12:
This is great news!

bonifaido, 2022-11-08:
This is real now: https://formulae.brew.sh/formula/wasm-micro-runtime so closing. Thank you for #1374 !

closed 2022-11-08 by bonifaido