wasm-micro-runtime/wasm-micro-runtime #1361: Different print between interpreter and AOT?

Zhangyx24, 2022-08-06:
## Steps to reproduce
[test-wat.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9274830/test-wat.txt)
[test-c.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9274831/test-c.txt)

The test.c is uoloaded as test-c.txt, and test.wat is uploaded as test-wat.txt.

（1）Interpreter
`cd build`
`cmake ..`
`make`
`./test`
（2）AoT
`cd AOTbuild`
`cmake -DWAMR_BUILD_INTERP=0 -DWAMR_BUILD_AOT=1 ..`
`make`
`./test`

## Environment
macOS 10 and ubuntu 20.04
wasmer 2.3.0

## Expected result
Initializing...
Loading binary...
Compiling module...
Instantiating module...
Extracting export...
Export size : 1
Shutting down...
Done.

## Actual result
The (1) mode could print the correct result, however the (2) mode didn't.

#### In macOS
mode (2) print:
Initializing...
Loading binary...
Compiling module...
Instantiating module...
Extracting export...
> Get 0 exports!
Shutting down...
Done.

The export is 1, however in macOS mode (2) print 0 export

#### In Ubuntu
mode (2) print:
Initializing...
Loading binary...
> Error loading module!

lum1n0us, 2022-08-09:
> Error loading module!

comes from `test.c`. It represents that "test.wasm" is not in the current directory.
``` c
  // Load binary.
  printf("Loading binary...\n");
#if WASM_ENABLE_AOT != 0 && WASM_ENABLE_INTERP == 0
  FILE* file = fopen("test.aot", "rb");
#else
  FILE* file = fopen("test.wasm", "rb");
#endif
  if (!file) {
    printf("> Error loading module!\n");
    return 1;
  }
````

Zhangyx24, 2022-08-09:
I have uploaded the test.wat, in your project, the conversion of wat to wasm file is executed in CMakeLists.txt, it means when execute the command `make`, your project will convert the test.wat to test.wasm automatically.

lum1n0us, 2022-08-10:
Yes. `cmake` will generate .wasm. I am not suggesting the "test.wasm" is not created. If so, the error will fail the build command.

*Error loading module!* suggests `FILE* file = fopen("test.wasm", "rb");` is failed. The most common reason of `fopen` failures is *can not find such a file* which in our case is *test.wasm*. Giving by the path of *test.wasm* is the current directory where the *test* binary is, 
- make sure "test" and "test.wasm" are in the same directory
- run "test" in the directory where it is located

Zhangyx24, 2022-08-10:
Thanks for your reply ! But  `test`, `test.wasm` and `test.wat` are in the same directory, and in my environment, WAMR could find test.wasm and load it, but get the wrong number of exports. The  loading module is not error in my test environment. I'm sorry that I'm not familiar with your environment.

(1)In interpreter mode, it prints the correct result
Initializing...
Loading binary...
Compiling module...
Instantiating module...
Extracting export...
Export size : 1
Shutting down...
Done.

(2)however, in AoT mode, it can only load the module, but get the wrong number of exports
Initializing...
Loading binary...
Compiling module...
Instantiating module...
Extracting export...
> Get 0 exports!
Shutting down...
Done.

lum1n0us, 2022-08-10:
In AoT mode, need to check "test.aot" instead of "test.wasm".

Zhangyx24, 2022-08-10:
Thanks a lot for your patience and answer !

closed 2022-08-10 by Zhangyx24