wasm-micro-runtime/wasm-micro-runtime #1362: Fail to get the correct result when calling host function ?

Zhangyx24, 2022-08-07:
## Steps to reproduce
[test-wat.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9276507/test-wat.txt)
[test-c.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9276508/test-c.txt)
test.c is uploaded as test-c.txt, and test.wat is uploaded as test-wat.txt.

(1) interpreter
`cd build`
`cmake ..`
`make`
`./test`

(2)AoT
`cd AOTbuild`
`cmake -DWAMR_BUILD_INTERP=0 -DWAMR_BUILD_AOT=1 ..`
`make`
`./test`

## Expected result
sum_i32_host Results: 3
sum_i32_host Results: -656906904
sum_i32_host Results: 2124157492

## Autual result
#### (1)
Initializing...
Loading binary...
Compiling module...
Creating callback...
Instantiating module...
Extracting export...
Export size : 1
Calling test ...
_i32_host Results: -392881592
#### (2)
Initializing...
Loading binary...
Compiling module...
Creating callback...
Instantiating module...
Extracting export...
Export size : 1
Calling test ...
_i32_host Results: -368068552
unhandled SIGSEGV, si_addr: 0x1e

The result is random.

yamt, 2022-08-08:
i guess you should enable compiler warnings.

* args/results should be wasm_val_vec_t, not wasm_val_t
* `printf("sum_i32_host Results: %d\n" + s);`: replace `+` with `,`
* i guess your callback function should return `NULL`.

Zhangyx24, 2022-08-09:
I modify the test.c:

`#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <inttypes.h>

#include "wasm_c_api.h"

#define own

void sum_i32(
    const wasm_val_vec_t* args, wasm_val_vec_t* results
) {
    int x = (args->data[0]).of.i32;
    int y = (args->data[1]).of.i32;
    int s = x + y;
    printf("sum_i32_host Results: %d\n", s);
}

int main(int argc, const char* argv[]) {
    // Initialize.
    printf("Initializing...\n");
    wasm_engine_t* engine = wasm_engine_new();
    wasm_store_t* store = wasm_store_new(engine);

    // Load binary.
    printf("Loading binary...\n");
#if WASM_ENABLE_AOT != 0 && WASM_ENABLE_INTERP == 0
    FILE* file = fopen("test.aot", "rb");
#else
    FILE* file = fopen("test.wasm", "rb");
#endif
    if (!file) {
        printf("> Error loading module!\n");
        return 1;
    }

    int ret = fseek(file, 0L, SEEK_END);
    if (ret == -1) {
        printf("> Error loading module!\n");
        fclose(file);
        return 1;
    }

    long file_size = ftell(file);
    if (file_size == -1) {
        printf("> Error loading module!\n");
        fclose(file);
        return 1;
    }

    ret = fseek(file, 0L, SEEK_SET);
    if (ret == -1) {
        printf("> Error loading module!\n");
        fclose(file);
        return 1;
    }

    wasm_byte_vec_t binary;
    wasm_byte_vec_new_uninitialized(&binary, file_size);
    if (fread(binary.data, file_size, 1, file) != 1) {
        printf("> Error loading module!\n");
        fclose(file);
        return 1;
    }
    fclose(file);

    // Compile.
    printf("Compiling module...\n");
    own wasm_module_t* module = wasm_module_new(store, &binary);
    if (!module) {
        printf("> Error compiling module!\n");
        return 1;
    }

    wasm_byte_vec_delete(&binary);

    // Create external print functions.
    printf("Creating callback...\n");

    own wasm_functype_t* func_type = wasm_functype_new_1_1(wasm_valtype_new_i32(), wasm_valtype_new_i32());
    own wasm_func_t* func = wasm_func_new(store, func_type, sum_i32);

    wasm_functype_delete(func_type);

    // Instantiate.
    printf("Instantiating module...\n");
    wasm_extern_t* externs[] = {
        wasm_func_as_extern(func)
    };

    wasm_extern_vec_t imports = WASM_ARRAY_VEC(externs);
    own wasm_instance_t* instance =
        wasm_instance_new(store, module, &imports, NULL);
    if (!instance) {
        printf("> Error instantiating module!\n");
        return 1;
    }

    printf("Extracting export...\n");
    own wasm_extern_vec_t exports;
    wasm_instance_exports(instance, &exports);
    if (exports.size == 0) {
        printf("> Error accessing exports!\n");
        return 1;
    }else{
        printf("Export size : %zu\n", exports.size);
    }
    const wasm_func_t* func1 = wasm_extern_as_func(exports.data[0]);
    if (func1 == NULL) {
        printf("> Error accessing export!\n");
        return 1;
    }

    printf("Calling test ...\n");
    wasm_val_vec_t args = WASM_EMPTY_VEC;
    wasm_val_vec_t results = WASM_EMPTY_VEC;
    wasm_func_call(func1, &args, &results);

    // Shut down.
    printf("Shutting down...\n");
    wasm_module_delete(module);
    wasm_instance_delete(instance);
    wasm_func_delete(func);
    wasm_store_delete(store);
    wasm_engine_delete(engine);

    // All done.
    printf("Done.\n");
    return 0;
}`

But the actual result is :
Initializing...
Loading binary...
Compiling module...
Creating callback...
Instantiating module...
Extracting export...
Export size : 1
Calling test ...
sum_i32_host Results: 3
Segmentation fault (core dumped)

Zhangyx24, 2022-08-09:
I modify the test.c:

#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <inttypes.h>

#include "wasm_c_api.h"

#define own

void sum_i32(
    const wasm_val_vec_t* args, wasm_val_vec_t* results
) {
    int x = (args->data[0]).of.i32;
    int y = (args->data[1]).of.i32;
    int s = x + y;
    printf("sum_i32_host Results: %d\n", s);
}

int main(int argc, const char* argv[]) {
    // Initialize.
    printf("Initializing...\n");
    wasm_engine_t* engine = wasm_engine_new();
    wasm_store_t* store = wasm_store_new(engine);

    // Load binary.
    printf("Loading binary...\n");
#if WASM_ENABLE_AOT != 0 && WASM_ENABLE_INTERP == 0
    FILE* file = fopen("test-wamr526.aot", "rb");
#else
    FILE* file = fopen("test-wamr526.wasm", "rb");
#endif
    if (!file) {
        printf("> Error loading module!\n");
        return 1;
    }

    int ret = fseek(file, 0L, SEEK_END);
    if (ret == -1) {
        printf("> Error loading module!\n");
        fclose(file);
        return 1;
    }

    long file_size = ftell(file);
    if (file_size == -1) {
        printf("> Error loading module!\n");
        fclose(file);
        return 1;
    }

    ret = fseek(file, 0L, SEEK_SET);
    if (ret == -1) {
        printf("> Error loading module!\n");
        fclose(file);
        return 1;
    }

    wasm_byte_vec_t binary;
    wasm_byte_vec_new_uninitialized(&binary, file_size);
    if (fread(binary.data, file_size, 1, file) != 1) {
        printf("> Error loading module!\n");
        fclose(file);
        return 1;
    }
    fclose(file);

    // Compile.
    printf("Compiling module...\n");
    own wasm_module_t* module = wasm_module_new(store, &binary);
    if (!module) {
        printf("> Error compiling module!\n");
        return 1;
    }

    wasm_byte_vec_delete(&binary);

    // Create external print functions.
    printf("Creating callback...\n");

    own wasm_functype_t* func_type = wasm_functype_new_1_1(wasm_valtype_new_i32(), wasm_valtype_new_i32());
    own wasm_func_t* func = wasm_func_new(store, func_type, sum_i32);

    wasm_functype_delete(func_type);

    // Instantiate.
    printf("Instantiating module...\n");
    wasm_extern_t* externs[] = {
        wasm_func_as_extern(func)
    };

    wasm_extern_vec_t imports = WASM_ARRAY_VEC(externs);
    own wasm_instance_t* instance =
        wasm_instance_new(store, module, &imports, NULL);
    if (!instance) {
        printf("> Error instantiating module!\n");
        return 1;
    }

    printf("Extracting export...\n");
    own wasm_extern_vec_t exports;
    wasm_instance_exports(instance, &exports);
    if (exports.size == 0) {
        printf("> Error accessing exports!\n");
        return 1;
    }else{
        printf("Export size : %zu\n", exports.size);
    }
    const wasm_func_t* func1 = wasm_extern_as_func(exports.data[0]);
    if (func1 == NULL) {
        printf("> Error accessing export!\n");
        return 1;
    }

    printf("Calling test ...\n");
    wasm_val_vec_t args = WASM_EMPTY_VEC;
    wasm_val_vec_t results = WASM_EMPTY_VEC;
    wasm_func_call(func1, &args, &results);

    // Shut down.
    printf("Shutting down...\n");
    wasm_module_delete(module);
    wasm_instance_delete(instance);
    wasm_func_delete(func);
    wasm_store_delete(store);
    wasm_engine_delete(engine);

    // All done.
    printf("Done.\n");
    return 0;
}

But the actual result is :
Initializing...
Loading binary...
Compiling module...
Creating callback...
Instantiating module...
Extracting export...
Export size : 1
Calling test ...
sum_i32_host Results: 3
Segmentation fault (core dumped)

lum1n0us, 2022-08-10:
need to patch

```c
    // it likes "func (param i32) (result i32)"
    own wasm_functype_t* func_type = wasm_functype_new_1_1(wasm_valtype_new_i32(), wasm_valtype_new_i32());
    own wasm_func_t* func = wasm_func_new(store, func_type, sum_i32);
```
==>
```c
    // it likes "func (param i32 i32)"
    own wasm_functype_t* func_type = wasm_functype_new_2_0(wasm_valtype_new_i32(), wasm_valtype_new_i32());
    own wasm_func_t* func = wasm_func_new(store, func_type, sum_i32);
```

Zhangyx24, 2022-08-10:
Thanks a lot !
I modify the code according to your reply, but still get the wrong result :

## In macOS
#### (1)interpreter mode
[image]

#### (2)AoT mode
[image]

## In Ubuntu
#### (1)interpreter mode
[image]

#### (2)AoT mode
[image]

lum1n0us, 2022-08-10:
My mistake on forgetting to mention that:

`void sum_i32(` should be `wasm_trap_t * sum_i32` and might want to add `return NULL;`  at the end.

Btw, please make sure to fix all compilation warnings.  Usually, they are helpful.

Zhangyx24, 2022-08-10:
Thanks a lot !

closed 2022-08-10 by Zhangyx24