wasm-micro-runtime/wasm-micro-runtime #1369: Incorrect result of allocation memory 0 ?

Zhangyx24, 2022-08-09:
## Steps to reproduce
[test2-wat.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9290632/test2-wat.txt)
[test1-c.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9290633/test1-c.txt)
[test2-c.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9290634/test2-c.txt)
[test1-wat.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9290636/test1-wat.txt)
 
test1.c is uploaded as test1-c.txt , so as test2.c.
test1.wat is uploaded as test1-wat.txt, so as test2.wat.

#### (1)interpreter
`cd build`
`cmake ..`
`make`
`./test1`
`./test2`
#### (2)AoT
`cd AOTbuild`
`cmake -DWAMR_BUILD_INTERP=0 -DWAMR_BUILD_AOT=1 ..`
`make`
`./test1`
`./test2`

Using interpreter or AoT mode to execute the `func1` and `func2` function in test1.wat and test2.wat respectively.

## Environment
macOS10.15 and ubuntu20.04
WAMR 05-18-2022

## Expected result
#### (1)result of executing test1.wat
equals i32 !
equals i32 !
equals i32 !
equals i32 !
#### (2)result of executing test2.wat
equals i32 !
equals i32 !
Not equals i32 !
Not equals i32 !
equals i32 !
equals i32 !
equals i64 !
equals f32 !
equals f64 !
equals i64 !
equals i64 !
equals i64 !
equals i32 !
equals i32 !
equals i32 !
equals i32 !
equals i64 !
equals i64 !
equals i64 !

## Actual result
#### (1)result of executing test1.wat

#### in interpreter mode
Not equals i32 !
Not equals i32 !
Not equals i32 !
Not equals i32 !

#### in AoT mode
Not equals i64 !
equals i32 !
Not equals i64 !
equals i32 !
Not equals i64 !
Not equals i64 !
Not equals i64 !
Not equals i64 !
Not equals i64 !
equals i32 !
Not equals i64 !
equals i32 !
Not equals i64 !
Not equals i64 !
Not equals i64 !
Not equals i64 !

#### (2)result of executing test2.wat

#### in interpreter mode
Export size : 1
equals f64 !
equals f64 !
equals f64 !
equals f64 !
equals f64 !
equals f64 !
equals f32 !
equals i64 !
equals i32 !
equals f32 !
equals f32 !
equals f32 !
equals f64 !
equals f64 !
equals f64 !
equals f64 !
equals f32 !
equals f32 !
equals f32 !

#### in AoT mode
equals f32 !
equals f64 !
Not equals f32 !
equals f64 !
Not equals f32 !
Not equals f32 !
Not equals f32 !
equals f32 !
Export size : 1
equals f32 !
equals f64 !
Not equals f32 !
equals f64 !
Not equals f32 !
Not equals f32 !
Not equals f32 !
equals f32 !

wenyongh, 2022-08-19:
@Zhangyx24 I tested the cases, for test1.wast, the interpreter's result and AOT's result are the same and as expected.
For test2.wast, the result is not as expected because that there is app heap inserted into the linear memory by default, so the initial page count might be larger than expected, see:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/memory_tune.md#the-memory-model

If you don't want to insert the app heap, please use `wasm_instance_new_with_args` instead of `wasm_instance_new`, and set heap size to 0, e.g.:
```C
wasm_instance_new_with_args(store, module, &imports, NULL, 32768, 0)
```
Refer to:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/include/wasm_c_api.h#L579-L583

Zhangyx24, 2022-08-19:
Thanks a lot! Now it works !