wasm-micro-runtime/wasm-micro-runtime #1389: Pass 12 characters values into WASM program written in C generates garbage character on the tail

cosmo0920, 2022-08-16:
## Step to replicate

1. Write WASM program runtime as:

```cpp
#include <array>
#include <fstream>
#include <iostream>
#include <wasm_export.h>

enum class Errors {
    INITIALIZE      = -10,
    LOAD            = -20,
    INSTANTIATE     = -30,
    INSTANTIATE_ENV = -40,
    FIND_FUNCTION   = -50,
    RUNNING         = -60,
};

int main(int argc, char* argv[]) {
  char *buffer;
  std::array<char, 128> error_buf;
  uint32_t size, stack_size = 8092, heap_size = 8092;

  auto wasm_initialized = wasm_runtime_init();
  const std::string func_name("say_hello");

  if (!wasm_initialized) {
    std::cerr << "Failed to initialize WASM!" << std::endl;
    return static_cast<int>(Errors::INITIALIZE);
  }

  std::ifstream bytecode_file("c_say_hello.wasm");

  std::string bytecode(
                       (std::istreambuf_iterator<char>(bytecode_file)),
                       std::istreambuf_iterator<char>());

  auto module = wasm_runtime_load(reinterpret_cast<uint8_t*>(const_cast<char*>(bytecode.data())), bytecode.size(), error_buf.data(), error_buf.size());

  if (!module) {
    std::cerr << "Failed to load bytecode! " << error_buf.data() << std::endl;
    return static_cast<int>(Errors::LOAD);
  }

  auto module_inst = wasm_runtime_instantiate(module, stack_size, heap_size, error_buf.data(), error_buf.size());

  if (!module_inst) {
    std::cerr << "Failed to instantiate module!" << error_buf.data()  << std::endl;
    return static_cast<int>(Errors::INSTANTIATE);
  }

  auto exec_env = wasm_runtime_create_exec_env(module_inst, stack_size);

  if (!exec_env) {
    std::cerr << "Failed to instantiate module!" << std::endl;
    return static_cast<int>(Errors::INSTANTIATE_ENV);
  }

  auto func = wasm_runtime_lookup_function(module_inst, func_name.c_str(), nullptr);

  if (!func) {
    std::cerr << "Failed to find \"" << func_name << "\" function!" << std::endl;
    return static_cast<int>(Errors::FIND_FUNCTION);
  }

  std::string arg("dummy.locals");
  auto wasm_buf = wasm_runtime_module_dup_data(module_inst, arg.data(), arg.size());
  std::array<uint32_t, 2> func_args = {wasm_buf, static_cast<unsigned int>(arg.size())};

  auto call_succeeded = wasm_runtime_call_wasm(exec_env, func, func_args.size(), func_args.data());

  if (!call_succeeded) {
    const char *exception = wasm_runtime_get_exception(module_inst);
    if (exception) {
      std::cerr << "Got exception running wasm code: " << exception << std::endl;
    }
    std::cerr << "Failed to call \"hello\"!" << std::endl;
    return static_cast<int>(Errors::RUNNING);
  }

  auto func_rv = wasm_runtime_addr_app_to_native(module_inst, func_args[0]);

  std::cout << "WASM function \"" << func_name << "\" returned: " << std::string(static_cast<char*>(func_rv)) << std::endl;

  wasm_runtime_module_free(module_inst, wasm_buf);

  return 0;
}
```
2. Build WASM runtime embedded executable

```console
$ g++ -I /path/to/wasm-micro-runtime/core/iwasm/include runtime/runtime.cpp -o wasm_runtime -L /path/to/wasm-micro-runtime/product-mini/platforms/linux/ -lvmlib -lpthread -std=c++11

```
3. Write WASM Program from C

```c
#include <stdio.h>
#include <stdint.h>
#include <stdlib.h>

char* say_hello(char* buf, int size)
{
    char *buffer;
    buffer = malloc(1024);
    if (!buffer) {
        printf("malloc buf failed\n");
        return NULL;
    }

    sprintf(buffer, "Hello, %s from C!", buf);
    return buffer;
}
```

4. Build WASM with WASI SDK

```console
$ /opt/wasi-sdk/bin/clang -O3 -nostdlib \
        -z stack-size=8192 -Wl,--initial-memory=65536 \
        -o c_say_hello.wasm wasm/c/main.c \
        -Wl,--export=__heap_base -Wl,--export=__data_end -Wl,--export=say_hello \
        -Wl,--no-entry -Wl,--strip-all -Wl,--allow-undefined
```

Then, I got a garbage character added string from WASM:

```console
$ WASM function "say_hello" returned: Hello, dummy.locals� from C!
```

## Environment

```console
$ /opt/wasi-sdk/bin/clang --version
clang version 14.0.4 (https://github.com/llvm/llvm-project 29f1039a7285a5c3a9c353d054140bf2556d4c4d)
Target: wasm32-unknown-wasi
Thread model: posix
InstalledDir: /opt/wasi-sdk/bin
```
```console
$ cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.4 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.4 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

## Additional context

Using tinygo to create WASM program, this issue is never occurred.
This issue is only occurred by passing 12 characters string values.

wenyongh, 2022-08-17:
Hi, this is because that you pass invalid size to wasm_runtime_module_dup_data:
```
  std::string arg("dummy.locals");
  auto wasm_buf = wasm_runtime_module_dup_data(module_inst, arg.data(), arg.size());
```
Note that here `arg.size()` is 12, it doesn't include the terminated character '\0', please use `arg.size() + 1` instead.

cosmo0920, 2022-08-17:
Oh, I got it. Thanks for the explanation. I'd forgotten to add for the null terminators' length.

cosmo0920, 2022-08-17:
Now, this issue is solved. Thanks for your help!!! :smile:

closed 2022-08-17 by cosmo0920