wasm-micro-runtime/wasm-micro-runtime #1399: error:WASM module load failed: invalid section id

weiqingtangx, 2022-08-19:
Hi guys, I use wasi-sdk16.0 to compile my c++ source code, as below:
```
#include <stdio.h>
#include <iostream>

int main(int argc, char **argv) {
    std::cout << "Hello world!"  << std::endl;
    std::string str("1234");
    std::cout << "Hi str:" << str << std::endl;
    
    const char *s = str.c_str();
    printf("str: %s, size: %d\n", s, (int)str.size());
    return 0;
}
```
and built it with wasi-sdk 16.0

```
/opt/wasi-sdk/bin/clang++ -O3 \
    -z stack-size=4096 -Wl,--initial-memory=131072 \
    -Wl,--export=malloc -Wl,--export=free \
    -o hello.wasm hello.cpp
```

then run it
```
./iwasm hello.wasm
```

get error 
```
error:WASM module load failed: invalid section id
```
Could you please tell me what's wrong with it? Thanks.

wenyongh, 2022-08-19:
@weiqingtangx Please enable the multi-thread feature when building iwasm: `cmake .. -DWAMR_BUILD_LIB_PTHREAD=1`

weiqingtangx, 2022-08-22:
I use ```cmake .. -DWAMR_BUILD_LIB_PTHREAD=1``` flag to build iwasm, then run hello.wasm, new errors occur:
```
(base) ➜  build git:(main) ✗ ./iwasm ~/Desktop/hello.wasm 
[23:43:39:977 - F97600]: warning: failed to link import function (wasi_snapshot_preview1, args_get)
[23:43:39:019 - F97600]: warning: failed to link import function (wasi_snapshot_preview1, args_sizes_get)
[23:43:39:025 - F97600]: warning: failed to link import function (wasi_snapshot_preview1, environ_get)
[23:43:39:029 - F97600]: warning: failed to link import function (wasi_snapshot_preview1, environ_sizes_get)
[23:43:39:033 - F97600]: warning: failed to link import function (wasi_snapshot_preview1, fd_close)
[23:43:39:037 - F97600]: warning: failed to link import function (wasi_snapshot_preview1, fd_fdstat_get)
[23:43:39:041 - F97600]: warning: failed to link import function (wasi_snapshot_preview1, fd_read)
[23:43:39:045 - F97600]: warning: failed to link import function (wasi_snapshot_preview1, fd_seek)
[23:43:39:049 - F97600]: warning: failed to link import function (wasi_snapshot_preview1, fd_write)
[23:43:39:053 - F97600]: warning: failed to link import function (wasi_snapshot_preview1, proc_exit)
Exception: lookup main function failed
```

weiqingtangx, 2022-08-22:
This is my wasm file, maybe you could use it for debugging.
[hello.wasm.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/9392988/hello.wasm.zip)

wenyongh, 2022-08-22:
@weiqingtangx Please build iwasm with `-DWAMR_BUILD_LIBC_WASI=1`, but not sure why it doesn't enabled in your compilation, by default it is enabled in darwin platform.

weiqingtangx, 2022-08-23:
Building iwasm with two flags  ```-DWAMR_BUILD_LIBC_WASI=1 -DWAMR_BUILD_LIB_PTHREAD=1``` could resolve this problem. Thank you.

weiqingtangx, 2022-08-23:
@wenyongh My demo code doesn't use any pthread, why should I use ```-DWAMR_BUILD_LIB_PTHREAD=1``` flag to build?

wenyongh, 2022-08-23:
@weiqingtangx try using `-DWAMR_BUILD_BULK_MEMORY=1` instead of `-DWAMR_BUILD_LIB_PTHREAD=1`.

weiqingtangx, 2022-08-23:
Got that. I checked wasi-sdk 16 release document, it used bulk memory. Thank you.
[image]

closed 2022-08-23 by weiqingtangx