wasm-micro-runtime/wasm-micro-runtime #1400: Un expected behavior in the instruction `memory.grow`

erxiaozhou, 2022-08-20:
I have found some unexpected behaviors when executing a test case with the instruction `memory.grow`.

# Build commands
I compile the code with commit id 653b926d53dc010fed8685145c2869fdcb4ffdc5 and the code with commit id ccd664b81eefaf256afeccf83c26236e10cdc57b.
Platform: Ubuntu 20.04
CPU: amd64 
compile:
```shell
export CC=/usr/bin/clang-14
export CXX=/usr/bin/clang++-14
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Debug -DWAMR_BUILD_FAST_INTERP=0 -Bbuild
cd build
make -j40
```
execute:
```shell
iwasm -f _start grow_poc.wasm
```

[grow_poc.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/9386429/grow_poc.zip)

# Expected behavior:
After executing memory.grow, the expected memory size is 125 and all bytes in the memory are zero.

# Actual behavior:
On the code with commit id 653b926d53dc010fed8685145c2869fdcb4ffdc5 and the code with commit id ccd664b81eefaf256afeccf83c26236e10cdc57b, after executing memory.grow, the memory size is 126.
By the way, after executing memory.grow, there are some non-zero bytes in the memory (code with commit id 653b926d53dc010fed8685145c2869fdcb4ffdc5), which is unexpected behavior. I have not checked whether there are non-zero bytes on the code with commit id ccd664b81eefaf256afeccf83c26236e10cdc57b. Maybe this unexpected behavior still exists.

Thank you.

erxiaozhou, 2022-08-20:
It seems that the default memory size is 2?

erxiaozhou, 2022-08-20:
It seems that the default memory size is 2 in iwasm?

xujuntwt95329, 2022-08-20:
Hi, we insert an additional `app heap` to the linear memory by default, so the initial page count might be larger than expected, see [memory_model](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/memory_tune.md#the-memory-model)

You can use `--heap-size=0` to avoid this:

``` bash
iwasm --heap-size=0 -f _start grow_poc.wasm
```

erxiaozhou, 2022-08-20:
> Hi, we insert an additional `app heap` to the linear memory by default, so the initial page count might be larger than expected, see [memory_model](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/memory_tune.md#the-memory-model)
> 
> You can use `--heap-size=0` to avoid this:
> 
> ```shell
> iwasm --heap-size=0 -f _start grow_poc.wasm
> ```

Thank you very much! `--heap-size=0` can lead to my expected output.
By the way, according to the specification of WASM, I think the data in the `app heap` should be initialized as zero. However, there are some non-zero bytes in the memory after executing the test case. I wonder whether this is an expected behavior. Thank you!

xujuntwt95329, 2022-08-22:
`app heap` is managed by our own allocator, so there are some data structures inside the `app_heap`, so you may encounter some non-zero bytes. But other spaces are all zero-initialized.

erxiaozhou, 2022-08-22:
> `app heap` is managed by our own allocator, so there are some data structures inside the `app_heap`, so you may encounter some non-zero bytes. But other spaces are all zero-initialized.

Get it. Thank you for your reply！

closed 2022-09-15 by wenyongh