wasm-micro-runtime/wasm-micro-runtime #1414: About the code in simd_conversions.c

sk-takeda, 2022-08-25:
Hi,

I'm investigating the "simd" option.
I think there is a problem with the code below

https://github.com/bytecodealliance/wasm-micro-runtime/blob/903a27aff208200311118d41952ad526a68815aa/core/iwasm/compilation/simd/simd_conversions.c#L138

Regarding the if statement above, I think "=" is correct instead of "==".
Please confirm.

wenyongh, 2022-08-30:
@sk-takeda Thanks a lot! We fixed it in #1428.

sk-takeda, 2022-08-31:
Thank you for your response and code fix.

I have an additional question.
It's a modified code, but I think it's the code used for simds other than x86_64.
From the documentation and past issues, I think simd only supports x86_64.
The wamrc help message is as follows.
"currently 128-bit SIMD is only supported for x86-64 target"
Does simd support architectures other than x86_64?

wenyongh, 2022-08-31:
Thanks, this should be a typo, simd is also supported for aarch64, ref to PR:
[ https://github.com/bytecodealliance/wasm-micro-runtime/pull/610](https://github.com/bytecodealliance/wasm-micro-runtime/pull/610)

sk-takeda, 2022-09-08:
Thank you.
I created an aot application on aarch64 using the modified one. The following log is output when building with wamrc.
If you do not set -target=aarch64 in WAMRC, this log will not be output.
Is the simd generated normally?
In addition, AOT is generated, and it seems that there is no problem in operation.

-----------------------
user1@1639bb4b42e8:/workspace$ /opt/wasm-micro-runtime/wamr-compiler/build/wamrc --enable-dump-call-stack --target=aarch64 -o /workspace/sysroot/device-was
m-dev/aarch64/samples/motion_detection.aot /workspace/sysroot/device-wasm-dev/wasm32/samples/motion_detection.wasm
Create AoT compiler with:
  target:        aarch64v8
  target cpu:
  cpu features:
  opt level:     3
  size level:    3
  output format: AoT file
Call parameter type does not match function signature!
  %vec1 = bitcast <2 x i64> %and33 to <4 x i32>
 <8 x i16>  %call = call <8 x i16> @llvm.umin.v8i16(<4 x i32> %vec1, <4 x i32> <i32 65535, i32 65535, i32 65535, i32 65535>)
Call parameter type does not match function signature!
<4 x i32> zeroinitializer
 <8 x i16>  %call34 = call <8 x i16> @llvm.umax.v8i16(<8 x i16> %call, <4 x i32> zeroinitializer)
Call parameter type does not match function signature!
  %vec2 = bitcast <2 x i64> %"local10#" to <4 x i32>
 <8 x i16>  %call35 = call <8 x i16> @llvm.umin.v8i16(<4 x i32> %vec2, <4 x i32> <i32 65535, i32 65535, i32 65535, i32 65535>)
Call parameter type does not match function signature!
<4 x i32> zeroinitializer
 <8 x i16>  %call36 = call <8 x i16> @llvm.umax.v8i16(<8 x i16> %call35, <4 x i32> zeroinitializer)
DestTy too big for Trunc
  %vec1_trunc = trunc <8 x i16> %call34 to <4 x i16>
DestTy too big for Trunc
  %vec2_trunc = trunc <8 x i16> %call36 to <4 x i16>
Call parameter type does not match function signature!
  %vec169 = bitcast <2 x i64> %and66 to <4 x i32>
 <8 x i16>  %call70 = call <8 x i16> @llvm.umin.v8i16(<4 x i32> %vec169, <4 x i32> <i32 65535, i32 65535, i32 65535, i32 65535>)
Call parameter type does not match function signature!
<4 x i32> zeroinitializer
 <8 x i16>  %call71 = call <8 x i16> @llvm.umax.v8i16(<8 x i16> %call70, <4 x i32> zeroinitializer)
Call parameter type does not match function signature!
  %vec268 = bitcast <2 x i64> %"local8#67" to <4 x i32>
 <8 x i16>  %call72 = call <8 x i16> @llvm.umin.v8i16(<4 x i32> %vec268, <4 x i32> <i32 65535, i32 65535, i32 65535, i32 65535>)
Call parameter type does not match function signature!
<4 x i32> zeroinitializer
 <8 x i16>  %call73 = call <8 x i16> @llvm.umax.v8i16(<8 x i16> %call72, <4 x i32> zeroinitializer)
DestTy too big for Trunc
  %vec1_trunc74 = trunc <8 x i16> %call71 to <4 x i16>
DestTy too big for Trunc
  %vec2_trunc75 = trunc <8 x i16> %call73 to <4 x i16>
Call parameter type does not match function signature!
  %vec1115 = bitcast <2 x i64> %and112 to <4 x i32>
 <8 x i16>  %call116 = call <8 x i16> @llvm.umin.v8i16(<4 x i32> %vec1115, <4 x i32> <i32 65535, i32 65535, i32 65535, i32 65535>)
Call parameter type does not match function signature!
<4 x i32> zeroinitializer
 <8 x i16>  %call117 = call <8 x i16> @llvm.umax.v8i16(<8 x i16> %call116, <4 x i32> zeroinitializer)
Call parameter type does not match function signature!
  %vec2114 = bitcast <2 x i64> %"local8#113" to <4 x i32>
 <8 x i16>  %call118 = call <8 x i16> @llvm.umin.v8i16(<4 x i32> %vec2114, <4 x i32> <i32 65535, i32 65535, i32 65535, i32 65535>)
Call parameter type does not match function signature!
<4 x i32> zeroinitializer
 <8 x i16>  %call119 = call <8 x i16> @llvm.umax.v8i16(<8 x i16> %call118, <4 x i32> zeroinitializer)
DestTy too big for Trunc
  %vec1_trunc120 = trunc <8 x i16> %call117 to <4 x i16>
DestTy too big for Trunc
  %vec2_trunc121 = trunc <8 x i16> %call119 to <4 x i16>
Call parameter type does not match function signature!
  %vec1 = bitcast <2 x i64> %and33 to <4 x i32>
 <8 x i16>  %call = call <8 x i16> @llvm.umin.v8i16(<4 x i32> %vec1, <4 x i32> <i32 65535, i32 65535, i32 65535, i32 65535>)
Call parameter type does not match function signature!
<4 x i32> zeroinitializer
 <8 x i16>  %call34 = call <8 x i16> @llvm.umax.v8i16(<8 x i16> %call, <4 x i32> zeroinitializer)
[... 802 lines cut ...]
 <8 x i16>  %call2123 = call <8 x i16> @llvm.umin.v8i16(<4 x i32> %vec22119, <4 x i32> <i32 65535, i32 65535, i32 65535, i32 65535>)
Call parameter type does not match function signature!
<4 x i32> zeroinitializer
 <8 x i16>  %call2124 = call <8 x i16> @llvm.umax.v8i16(<8 x i16> %call2123, <4 x i32> zeroinitializer)
DestTy too big for Trunc
  %vec1_trunc2125 = trunc <8 x i16> %call2122 to <4 x i16>
DestTy too big for Trunc
  %vec2_trunc2126 = trunc <8 x i16> %call2124 to <4 x i16>
Call parameter type does not match function signature!
  %vec12158 = bitcast <2 x i64> %and2155 to <4 x i32>
 <8 x i16>  %call2159 = call <8 x i16> @llvm.umin.v8i16(<4 x i32> %vec12158, <4 x i32> <i32 65535, i32 65535, i32 65535, i32 65535>)
Call parameter type does not match function signature!
<4 x i32> zeroinitializer
 <8 x i16>  %call2160 = call <8 x i16> @llvm.umax.v8i16(<8 x i16> %call2159, <4 x i32> zeroinitializer)
Call parameter type does not match function signature!
  %vec22157 = bitcast <2 x i64> %"local29#2156" to <4 x i32>
 <8 x i16>  %call2161 = call <8 x i16> @llvm.umin.v8i16(<4 x i32> %vec22157, <4 x i32> <i32 65535, i32 65535, i32 65535, i32 65535>)
Call parameter type does not match function signature!
<4 x i32> zeroinitializer
 <8 x i16>  %call2162 = call <8 x i16> @llvm.umax.v8i16(<8 x i16> %call2161, <4 x i32> zeroinitializer)
DestTy too big for Trunc
  %vec1_trunc2163 = trunc <8 x i16> %call2160 to <4 x i16>
DestTy too big for Trunc
  %vec2_trunc2164 = trunc <8 x i16> %call2162 to <4 x i16>
Compile success, file /workspace/sysroot/device-wasm-dev/aarch64/samples/motion_detection.aot was generated.