wasm-micro-runtime/wasm-micro-runtime #1418: Project provided and bundled Dockerfiles generate flood of errors and warnings with hadolint

cosmo0920, 2022-08-25:
## Step To Reproduce

1. Install hadolint 2.10.0
2. execute hadolint for Dockerfiles
3. Boom. :boom:

### hadolint version

```console
$ hadolint --version
Haskell Dockerfile Linter 2.10.0
```

## Result

```console
$ git ls-files --exclude="*Dockerfile*" --cached --ignored | xargs hadolint
.devcontainer/Dockerfile:12 DL3027 warning: Do not use apt as it is meant to be a end-user tool, use apt-get or apt-cache instead
.devcontainer/Dockerfile:22 DL4006 warning: Set the SHELL option -o pipefail before RUN with a pipe in it. If you are using /bin/sh in an alpine image or if your shell is symlinked to busybox then consider explicitly setting your SHELL to /bin/ash, or disable this check
.devcontainer/Dockerfile:22 DL3047 info: Avoid use of wget without progress bar. Use `wget --progress=dot:giga <url>`.Or consider using `-q` or `-nv` (shorthands for `--quiet` or `--no-verbose`).
.devcontainer/Dockerfile:22 DL3027 warning: Do not use apt as it is meant to be a end-user tool, use apt-get or apt-cache instead
.devcontainer/Dockerfile:31 DL3003 warning: Use WORKDIR to switch to a directory
.devcontainer/Dockerfile:33 DL3003 warning: Use WORKDIR to switch to a directory
.devcontainer/Dockerfile:42 DL3047 info: Avoid use of wget without progress bar. Use `wget --progress=dot:giga <url>`.Or consider using `-q` or `-nv` (shorthands for `--quiet` or `--no-verbose`).
.devcontainer/Dockerfile:50 DL3047 info: Avoid use of wget without progress bar. Use `wget --progress=dot:giga <url>`.Or consider using `-q` or `-nv` (shorthands for `--quiet` or `--no-verbose`).
.devcontainer/Dockerfile:59 DL3059 info: Multiple consecutive `RUN` instructions. Consider consolidation.
.devcontainer/Dockerfile:59 DL3047 info: Avoid use of wget without progress bar. Use `wget --progress=dot:giga <url>`.Or consider using `-q` or `-nv` (shorthands for `--quiet` or `--no-verbose`).
.devcontainer/Dockerfile:65 DL3003 warning: Use WORKDIR to switch to a directory
.devcontainer/Dockerfile:68 DL3047 info: Avoid use of wget without progress bar. Use `wget --progress=dot:giga <url>`.Or consider using `-q` or `-nv` (shorthands for `--quiet` or `--no-verbose`).
.devcontainer/Dockerfile:68 DL3003 warning: Use WORKDIR to switch to a directory
.devcontainer/Dockerfile:72 DL3059 info: Multiple consecutive `RUN` instructions. Consider consolidation.
.devcontainer/Dockerfile:79 DL3027 warning: Do not use apt as it is meant to be a end-user tool, use apt-get or apt-cache instead
.devcontainer/Dockerfile:80 DL3013 warning: Pin versions in pip. Instead of `pip install <package>` use `pip install <package>==<version>` or `pip install --requirement <requirements file>`
.devcontainer/Dockerfile:80 DL3042 warning: Avoid use of cache directory with pip. Use `pip install --no-cache-dir <package>`
.devcontainer/Dockerfile:84 DL3059 info: Multiple consecutive `RUN` instructions. Consider consolidation.
.devcontainer/Dockerfile:84 DL3013 warning: Pin versions in pip. Instead of `pip install <package>` use `pip install <package>==<version>` or `pip install --requirement <requirements file>`
.devcontainer/Dockerfile:84 DL3042 warning: Avoid use of cache directory with pip. Use `pip install --no-cache-dir <package>`
.devcontainer/Dockerfile:92 DL3059 info: Multiple consecutive `RUN` instructions. Consider consolidation.
.devcontainer/Dockerfile:92 SC2028 info: echo may not expand escape sequences. Use printf.
ci/Dockerfile:12 DL3027 warning: Do not use apt as it is meant to be a end-user tool, use apt-get or apt-cache instead
ci/Dockerfile:22 DL4006 warning: Set the SHELL option -o pipefail before RUN with a pipe in it. If you are using /bin/sh in an alpine image or if your shell is symlinked to busybox then consider explicitly setting your SHELL to /bin/ash, or disable this check
ci/Dockerfile:22 DL3027 warning: Do not use apt as it is meant to be a end-user tool, use apt-get or apt-cache instead
ci/Dockerfile:22 DL3047 info: Avoid use of wget without progress bar. Use `wget --progress=dot:giga <url>`.Or consider using `-q` or `-nv` (shorthands for `--quiet` or `--no-verbose`).
ci/Dockerfile:31 DL3003 warning: Use WORKDIR to switch to a directory
ci/Dockerfile:33 DL3003 warning: Use WORKDIR to switch to a directory
ci/Dockerfile:42 DL3047 info: Avoid use of wget without progress bar. Use `wget --progress=dot:giga <url>`.Or consider using `-q` or `-nv` (shorthands for `--quiet` or `--no-verbose`).
ci/Dockerfile:43 DL3003 warning: Use WORKDIR to switch to a directory
ci/Dockerfile:52 DL3047 info: Avoid use of wget without progress bar. Use `wget --progress=dot:giga <url>`.Or consider using `-q` or `-nv` (shorthands for `--quiet` or `--no-verbose`).
ci/Dockerfile:60 DL3047 info: Avoid use of wget without progress bar. Use `wget --progress=dot:giga <url>`.Or consider using `-q` or `-nv` (shorthands for `--quiet` or `--no-verbose`).
ci/Dockerfile:69 DL3047 info: Avoid use of wget without progress bar. Use `wget --progress=dot:giga <url>`.Or consider using `-q` or `-nv` (shorthands for `--quiet` or `--no-verbose`).
ci/Dockerfile:69 DL3059 info: Multiple consecutive `RUN` instructions. Consider consolidation.
ci/Dockerfile:75 DL3027 warning: Do not use apt as it is meant to be a end-user tool, use apt-get or apt-cache instead
ci/Dockerfile:83 DL3059 info: Multiple consecutive `RUN` instructions. Consider consolidation.
ci/Dockerfile:83 SC2028 info: echo may not expand escape sequences. Use printf.
product-mini/platforms/zephyr/simple/Dockerfile:10 DL3027 warning: Do not use apt as it is meant to be a end-user tool, use apt-get or apt-cache instead
product-mini/platforms/zephyr/simple/Dockerfile:10 DL3005 error: Do not use apt-get dist-upgrade
product-mini/platforms/zephyr/simple/Dockerfile:10 DL3009 info: Delete the apt-get lists after installing something
product-mini/platforms/zephyr/simple/Dockerfile:18 DL3046 warning: `useradd` without flag `-l` and high UID will result in excessively large Image.
product-mini/platforms/zephyr/simple/Dockerfile:24 DL3013 warning: Pin versions in pip. Instead of `pip install <package>` use `pip install <package>==<version>` or `pip install --requirement <requirements file>`
product-mini/platforms/zephyr/simple/Dockerfile:24 DL3042 warning: Avoid use of cache directory with pip. Use `pip install --no-cache-dir <package>`
product-mini/platforms/zephyr/simple/Dockerfile:26 DL3003 warning: Use WORKDIR to switch to a directory
product-mini/platforms/zephyr/simple/Dockerfile:28 DL3042 warning: Avoid use of cache directory with pip. Use `pip install --no-cache-dir <package>`
test-tools/IoT-APP-Store-Demo/wasm_django/Dockerfile:4 DL3020 error: Use COPY instead of ADD for files and folders
test-tools/IoT-APP-Store-Demo/wasm_django/Dockerfile:6 DL3013 warning: Pin versions in pip. Instead of `pip install <package>` use `pip install <package>==<version>` or `pip install --requirement <requirements file>`
test-tools/IoT-APP-Store-Demo/wasm_django/Dockerfile:6 DL3042 warning: Avoid use of cache directory with pip. Use `pip install --no-cache-dir <package>`
test-tools/IoT-APP-Store-Demo/wasm_django/server/Dockerfile:4 DL3020 error: Use COPY instead of ADD for files and folders
test-tools/wamr-ide/WASM-Debug-Server/Docker/Dockerfile:11 DL3008 warning: Pin versions in apt get install. Instead of `apt-get install <package>` use `apt-get install <package>=<version>`
test-tools/wamr-ide/WASM-Debug-Server/Docker/Dockerfile:11 DL3047 info: Avoid use of wget without progress bar. Use `wget --progress=dot:giga <url>`.Or consider using `-q` or `-nv` (shorthands for `--quiet` or `--no-verbose`).
test-tools/wamr-ide/WASM-Debug-Server/Docker/Dockerfile:11 DL3015 info: Avoid additional packages by specifying `--no-install-recommends`
test-tools/wamr-ide/WASM-Debug-Server/Docker/Dockerfile:11 DL3014 warning: Use the `-y` switch to avoid manual input `apt-get -y install <package>`
test-tools/wamr-ide/WASM-Debug-Server/Docker/Dockerfile:19 DL3003 warning: Use WORKDIR to switch to a directory
test-tools/wamr-ide/WASM-Toolchain/Docker/Dockerfile:13 DL3047 info: Avoid use of wget without progress bar. Use `wget --progress=dot:giga <url>`.Or consider using `-q` or `-nv` (shorthands for `--quiet` or `--no-verbose`).
test-tools/wamr-ide/WASM-Toolchain/Docker/Dockerfile:13 DL3008 warning: Pin versions in apt get install. Instead of `apt-get install <package>` use `apt-get install <package>=<version>`
[... cut ...]
test-tools/wamr-ide/WASM-Toolchain/Docker/Dockerfile:68 DL3008 warning: Pin versions in apt get install. Instead of `apt-get install <package>` use `apt-get install <package>=<version>`
test-tools/wamr-ide/WASM-Toolchain/Docker/Dockerfile:68 DL3014 warning: Use the `-y` switch to avoid manual input `apt-get -y install <package>`
test-tools/wamr-ide/WASM-Toolchain/Docker/Dockerfile:68 DL3015 info: Avoid additional packages by specifying `--no-install-recommends`
```

## Expected results

Suppress errors and warnings at least.

TianlongLiang, 2022-09-22:
Hi, @cosmo0920 we have fixed all linter errors and infos, and most of the warnings except DL3008 and DL3013 because the unlikely nature of such conflict will occur for the package we use. Any other ideas or suggestions? We'd love to hear your feedback

cosmo0920, 2022-09-22:
Sure, I'll take a look. Thanks for your fixing.

cosmo0920, 2022-09-22:
I sent my thoughts as a PR: https://github.com/bytecodealliance/wasm-micro-runtime/pull/1511.

In our dockerfiles such as fluent-bit ones, we suppress linting errors for pinning versions:
https://github.com/fluent/fluent-bit/tree/master/packaging/distros

cosmo0920, 2022-09-24:
At 3693cbe54db6cee3fedccd0d5666ffa430bcfb0b, this issue is fixed. Thanks for your work and suggestions! :+1:

closed 2022-09-24 by cosmo0920