wasm-micro-runtime/wasm-micro-runtime #1439: Samples/basic could not be run with aot scenario

pengzeshan, 2022-08-31:
Environment: Ubuntu 22.04 with the kernel of 5.19.0-051900-generic #202207312230 SMP PREEMPT_DYNAMIC Sun Jul 31 22:34:11 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

Here are the steps to re-produce the scenario.
1. build the linux version with the following steps
"cd ~/workspace/open_source_projects/wasm-micro-runtime/product-mini/platforms/linux" 
"mkdir build", "cd build" and 
"cmake .. -DWAMR_BUILD_FAST_JIT=1"
make
2. build wamr with the following steps
cd wamr-compiler
./build_llvm.sh (or "./build_llvm_xtensa.sh" to support xtensa target)
mkdir build && cd build
cmake .. (or "cmake .. -DWAMR_BUILD_PLATFORM=darwin" for MacOS)
make
3. build testapp.aot:
cd wasm-micro-runtime/samples/basic
run ./build.sh 
run the cmd for generating testapp.aot:
/root/workspace/open_source_projects/wasm/wasm-micro-runtime/wamr-compiler/build/wamrc \
      -o ./out/wasm-apps/testapp.aot \
      ./out/wasm-apps/testapp.wasm
4. check testapp.aot with the cmd:
out/basic -f out/wasm-apps/testapp.aot

The error message is 
./run.sh: line 4: 614837 Illegal instruction     (core dumped) out/basic -f out/wasm-apps/testapp.aot

xujuntwt95329, 2022-09-06:
Hi, I followed your steps but can't reproduce the problem. My OS is Ubuntu 22.04, but kernel version is `5.15.0-46-generic`

Here is the aot file generated from my machine: 
[testapp.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/9493135/testapp.zip)

pengzeshan, 2022-09-06:
Hi, Xujun,
Thanks for taking time verifying the case. I just downloaded the attachment above and had a try:
./out/basic -f testapp.aot
Illegal instruction (core dumped)
I got the same error with the testapp.aot. 
It may be caused by the diff kernel versions as you noticed.
Thanks.
Damon