wasm-micro-runtime/wasm-micro-runtime #1463: Different behavior between other runtimes

erxiaozhou, 2022-09-08:
I use different runtimes to run the file, and get a different result on iwasm.

# Build commands
I compile the code with commit id da79e3e9b283272a63ad53b24310d88c640fbd8a.
Platform: Ubuntu 20.04
CPU: amd64 
compile:
```shell
export CC=/usr/bin/clang-14
export CXX=/usr/bin/clang++-14
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Debug -DWAMR_BUILD_FAST_INTERP=0 -Bbuild
cd build
make -j40
```
execute:
```shell
iwasm --heap-size=0 -f _start cp907_b1.wasm
```

[cp907_b1.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/9523519/cp907_b1.zip)

# Expected behavior:
On wasmtime, wasmer, wasmi, and wavm, the execution failed due to "unknown opcode"

# Actual behavior:
On the code with commit id da79e3e9b283272a63ad53b24310d88c640fbd8a , iwasm output:
```
0x3ff570a3:i32
```

Thank you.

wenyongh, 2022-09-08:
@erxiaozhou Thanks for reporting the issue! I uploaded PR #1464, please try again.

wenyongh, 2022-09-22:
Close this issue as it was fixed.

closed 2022-09-22 by wenyongh