wasm-micro-runtime/wasm-micro-runtime #1474: Unpected Behavior in iwasm

erxiaozhou, 2022-09-11:
I compile the code with commit id 75936680662ddc31af6763daa55a622422f9e2f5.
Platform: Ubuntu 20.04
CPU: amd64
compile:
```shell
export CC=/usr/bin/clang-14
export CXX=/usr/bin/clang++-14
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Debug -DWAMR_BUILD_FAST_INTERP=0 -Bbuild
cd build
make -j40
```

execute:
```shell
iwasm --heap-size=0 -f _start cp907_b2.wasm
```
[cp907_b2.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/9542758/cp907_b2.zip)

# Expected behavior:
It seems that there will be an exception indicating "unexpected data at the end of section".

# Actual behavior:
On the code with commit id 75936680662ddc31af6763daa55a622422f9e2f5, iwasm output:
```
0xd9cdb5a312ab0669:i64
```

Thank you.

wenyongh, 2022-09-13:
@erxiaozhou Thanks for reporting the issue, I fixed it in PR #1480, please try again.

wenyongh, 2022-09-22:
Close this issue as it was fixed.

closed 2022-09-22 by wenyongh