wasm-micro-runtime/wasm-micro-runtime #1476: Unexpected value on the stack

erxiaozhou, 2022-09-12:
# Build commands
I compile the code with commit id 75936680662ddc31af6763daa55a622422f9e2f5.
Platform: Ubuntu 20.04
CPU: amd64 
compile:
```shell
export CC=/usr/bin/clang-14
export CXX=/usr/bin/clang++-14
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Debug -DWAMR_BUILD_FAST_INTERP=0 -Bbuild
cd build
make -j40
```
execute:
```shell
iwasm --heap-size=0 -f _start cp907_b3.wasm
```

[cp907_b3.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/9549123/cp907_b3.zip)

# Expected behavior:
The value left on the stack is f4 fe 8e ff

# Actual behavior:
The value left on the stack is f4 fe ce ff

Thank you.

wenyongh, 2022-09-13:
@erxiaozhou Thanks for reporting the issue, I fixed it in PR #1480, please try again.

wenyongh, 2022-09-22:
Close this issue as it was fixed.

closed 2022-09-22 by wenyongh