wasm-micro-runtime/wasm-micro-runtime #1477: Unexpected value on the stack

erxiaozhou, 2022-09-12:
# Build commands
I compile the code with commit id 75936680662ddc31af6763daa55a622422f9e2f5.
Platform: Ubuntu 20.04
CPU: amd64 
compile:
```shell
export CC=/usr/bin/clang-14
export CXX=/usr/bin/clang++-14
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Debug -DWAMR_BUILD_FAST_INTERP=0 -Bbuild
cd build
make -j40
```
execute:
```shell
iwasm --heap-size=0 -f _start cp907_b4.wasm
```
[cp907_b4.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/9549360/cp907_b4.zip)

# Expected behavior:
The value left on the stack is 04 38 b3 7f

# Actual behavior:
The value left on the stack is 04 38 f3 7f

Maybe this unexpected behavior has a similar root cause to issue #1476.
Thank you.

wenyongh, 2022-09-13:
@erxiaozhou I tested in Ubuntu 20.04, x86-64, and cannot reproduce the issue. Could you please try modifying `fabs` below into `fabsf` in core/iwasm/interpreter/wasm_interpre_classic.c, and try again:
```
            HANDLE_OP(WASM_OP_F32_ABS)
            {
                DEF_OP_MATH(float32, F32, fabs);
                HANDLE_OP_END();
            }
```

erxiaozhou, 2022-09-13:
> @erxiaozhou I tested in Ubuntu 20.04, x86-64, and cannot reproduce the issue. Could you please try modifying `fabs` below into `fabsf` in core/iwasm/interpreter/wasm_interpre_classic.c, and try [...]

Thank you for your reply, modifying fabs to fabsf can fix this unexpected value.
Before fixing it, the value on frame_sp is
[image]
After modifying fabs to fabsf, the value on frame_sp is as expected.

[image]

wenyongh, 2022-09-13:
OK, I changed it to fabsf, and also applied similar changes to some other f32 operations, like ceilf, floorf, truncf, see #1480.

wenyongh, 2022-09-22:
Close this issue as it was fixed.

closed 2022-09-22 by wenyongh