wasm-micro-runtime/wasm-micro-runtime #1482: binary releases

yamt, 2022-09-14:
i think it makes sense for us to provide some binary assets for releases.
i guess we can start from wamrc, which takes long to build because of llvm.
how do you think?

wenyongh, 2022-09-15:
@yamt yes, it should be a good time to provide some binary releases, our suggestion is to provide: iwasm, wamrc, wamr-sdk, lldb and wamr-ide (vscode extension). Each of them has a binary for platform Ubuntu (maybe Ubuntu 20.04 and 22.04), Windows and MacOS. And these binary can be automatically generated by CI when we create a new WAMR release.

tinspin, 2022-09-20:
This would be great.

lum1n0us, 2022-09-30:
some optimization and feature-development related to this issue
- [ ] https://github.com/bytecodealliance/wasm-micro-runtime/issues/1537
- [ ] https://github.com/bytecodealliance/wasm-micro-runtime/issues/1545
- [ ] https://github.com/bytecodealliance/wasm-micro-runtime/issues/1524
- [ ] https://github.com/bytecodealliance/wasm-micro-runtime/issues/1510