wasm-micro-runtime/wasm-micro-runtime #1484: Platform ESP-IDF broken for WAMR 1.0.0 with ESP-IDF 4.4.2

Therdel, 2022-09-14:
With the most recent releases of WAMR and ESP-IDF, the product-mini build (and, probably, the whole esp-idf platform support) fails with a linker error.

The error occurs because the function `ftruncate` is implemented twice:
1. in the WAMR esp-idf platform support: [link (wamr1.0.0)](https://github.com/bytecodealliance/wasm-micro-runtime/blob/WAMR-1.0.0/core/shared/platform/esp-idf/espidf_platform.c#L231-L236)
2. in the ESP-IDF vfs component: [link (esp-idf4.4.2)](https://github.com/espressif/esp-idf/blob/v4.4.2/components/vfs/vfs.c#L1276-L1277)

Here's the relevant output:
> /home/.../.espressif/tools/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/wamr/libwamr.a(espidf_platform.c.obj): in function 'ftruncate':
> /home/.../dev/wamr_1-0-0/core/shared/platform/esp-idf/espidf_platform.c:233: multiple definition of 'ftruncate'; esp-idf/vfs/libvfs.a(vfs.c.obj):/home/.../esp/esp-idf/components/vfs/vfs.c:791: first defined here

I tested the following version combinations.
It seems to be a problem with exactly the respective most recent releases:
- WAMR 1.0.0 **fails** to build with ESP-IDF 4.4.2 (both the most recent releases)
- WAMR 1.0.0 **works** with ESP-IDF 4.4.1
- WAMR 1.0.0 **works** with ESP-IDF 4.3.3 & 4.3.4
- WAMR-05-18-2022 **works** with ESP-IDF 4.4.2

The relevant commits, I think, are:
(WAMR) [esp-idf: Make esp-idf support Libc WASI](https://github.com/bytecodealliance/wasm-micro-runtime/commit/071b8c25100aa7e9483e2cefcd491d28fb88468b)
(ESP-IDF) [fatfs: Add ftruncate api support for VFS and FAT-VFS ](https://github.com/espressif/esp-idf/commit/ebb9cc367021851a85a0567fc989429599bb2866)

dongsheng28849455, 2022-09-26:
hi, @Therdel, please check if the PR could fix the issue.
https://github.com/bytecodealliance/wasm-micro-runtime/pull/1520

Therdel, 2022-09-26:
It did (as far as I can tell).
I tested your WAMR-1.0.0 fix branch with both ESP-IDF 4.4.2 and 4.4.1 versions on an ESP32C3 - compilation and my application worked.

closed 2022-09-27 by wenyongh