wasm-micro-runtime/wasm-micro-runtime #1491: WASI implementation test automation

loganek, 2022-09-15:
I couldn't find any test suite in WAMR which tests end-to-end WASI interfaces. I thought it'd be good to have a set of tests (possibly in a form of standalone applications with assertions) that get executed on WAMR and execute various WASI methods (e.g. file operations etc.). I'm more than happy to work on details of that and share my ideas, but before that I'd like to know community thoughts, and whether there's already something like that planned?

JamesMenetrey, 2022-09-16:
Hello @loganek,

I think that would be a great idea to have some additional end-to-end WASI tests. To add some information regarding file operations, I have written some tests under the form of a sample that can be found in my recent pull request: #1489 (check [this file](https://github.com/bytecodealliance/wasm-micro-runtime/blob/c9053c7d153f2a56c2b6db6a79625fcfbb3700a0/samples/file/wasm-app/main.c)).

The location of such tests would also be a good discussion with the devs since I'm not sure the samples directory is the best location for them.

Cheers!

loganek, 2022-09-16:
Thanks @JamesMenetrey for the feedback. Whereas I'm yet to come up with a concrete proposal, I think it's probably better to keep sample apps and tests separate, as their purpose is slightly different. The purpose of sample apps is to demonstrate the usage of the APIs in a lightweight, possibly more a tutorial-style form, whereas tests are expected to check various edge cases, be exhaustive etc. 
I think there always will be an overlap here, but that's OK. Also, I think some of our sample apps are more like test apps, so once we have a separation, we can move them around.

lum1n0us, 2022-09-22:
Wamr is using a separate repo to maintain WASI test suites . It is forked from another open source project and kind of out-of-date, but still a good reference.

If want to reuse that, please let us know, and we will add your team into that repo. Or another brand new test framework is welcomed. In that case, it is free to create the new test framework in either one of both.

yamt, 2022-09-26:
> Wamr is using a separate repo to maintain WASI test suites . It is forked from another open source project and kind of out-of-date, but still a good reference.

i'm interested in the repo too.

> 
> If want to reuse that, please let us know, and we will add your team into that repo. Or another brand new test framework is welcomed. In that case, it is free to create the new test framework in either one of both.

i occasionally use a fork of wasm3 wasi tests for the purpose. its coverage is not great though.

wenyongh, 2022-09-27:
> > Wamr is using a separate repo to maintain WASI test suites . It is forked from another open source project and kind of out-of-date, but still a good reference.
> 
> i'm interested in the repo too.
> 
@yamt I have added you to the user list of the repo, could you please check the email and accept the invitation?

yamt, 2022-09-27:
> > > Wamr is using a separate repo to maintain WASI test suites . It is forked from another open source project and kind of out-of-date, but still a good reference.
> > 
> > 
> > i'm interested in the repo too.
> 
> @yamt I have added you to the user list of the repo, could you please check the email?

i accepted the invitation. thank you.

JamesMenetrey, 2022-09-27:
Hey! I would also be interested in getting access to this repository, please. :)

wenyongh, 2022-09-27:
@JamesMenetrey added :)

JamesMenetrey, 2022-09-27:
@wenyongh Great thanks!