wasm-micro-runtime/wasm-micro-runtime #1499: build_llvm.py question

nchamzn, 2022-09-16:
Hi,

Is there a strong reason for ``build_llvm.py`` being written in python?  Would you accept a pull request to convert this to a cmake script so it can be more easily integrated with the CMake in ``wamr-compiler``?

Thanks.

lum1n0us, 2022-09-18:
WAMR will use it to prepare llvm libraries compilation in CI and help to setup developers' daily environment.

nchamzn, 2022-09-26:
I understand the usage, I'm wondering why python is used rather than cmake

lum1n0us, 2022-09-27:
- more convenient and easy to maintain. 
- always thought LLVM libraries are a precondition like GCC and CMAKE. not something should be generated temporarily.

nchamzn, 2022-09-27:
As above though:

> WAMR will use it to prepare llvm libraries compilation in CI 

This suggests it is generated temporarily on CI at least and part of the build system?  Or do you bake the LLVM build in to your CI images?

The reason I ask is that I am integrating wamrc with a build system which does not have python available.  For now I copy/pasted the build definitions from build_llvm.py to my own cmake but it would be convenient to push a solution upstream so I don't need to maintain this.

lum1n0us, 2022-09-30:
Actually, at first, we used to use [`apt-get install llvm`](https://apt.llvm.org/) to prepare LLVM libraries. But went to `build_llvm.py` because of the unstable installation processing. What I am saying is, you don't need to compare LLVM libraries every time when compiling *wamrc* or *iwasm*.  It is better to think it as a programming tool or some development environment dependency.

You could upload "your own cmake", which I suppose is used to compile LLVM libraries only, to `build-scripts`.

nchamzn, 2022-11-22:
Thank you for the responses, I'll close this.

closed 2022-11-22 by nchamzn