wasm-micro-runtime/wasm-micro-runtime #1524: [CI Optimization] Remove `check_repo`

lum1n0us, 2022-09-27:
Remove [`check_repo`](https://github.com/bytecodealliance/wasm-micro-runtime/blob/8436e88a0718af8614714bacfdff4cf45958def5/.github/workflows/compilation_on_android_ubuntu.yml#L65)  and its output and all related references.

- [ ] compilation_on_android_ubuntu.yml  
- [ ] compilation_on_macos.yml  
- [ ] compilation_on_nuttx.yml  
- [ ] compilation_on_sgx.yml  
- [ ] compilation_on_windows.yml

TianlongLiang, 2022-09-28:
I removed it and found a way to disable jobs run on the forked repo by adding `if condition` for all jobs(but for testing purposes, comment it for now). I also made two jobs into independent reusable workflows, and so far, so good. But there is one thing worth noticing is that when passing the argument to called workflow, there are only three type are supported(string, number, boolean). Though there is a way to [walk around](https://github.com/community/community/discussions/11692) this problem and pass the type you want, it could become more tricky when input is complex

the current status can be seen in [this branch](https://github.com/TianlongLiang/wasm-micro-runtime/tree/reuseable_workflow )

lum1n0us, 2022-09-29:
Second thought, maybe run workflows on foked repo is not a bad idea. Developers can use it as a verification environment and are able to prepare a high quality PR. So how about keeping enable jobs run on foked repos?

lum1n0us, 2022-09-29:
`build_llvm.py --project clang lldb` is about to generate wamr-lldb binary. I guess we don't need it when building llvm libraries for *iwasm* and *wamrc*.

we can use another workflow, like * _build_wamr_lldb.yml* to handle wamr-lldb binary specificaly.

lum1n0us, 2022-09-29:
> Second thought, maybe run workflows on foked repo is not a bad idea. Developers can use it as a verification environment and are able to prepare a high quality PR. So how about keeping enable jobs run on foked repos?

But we need to make sure run the release job only on *bytecodealliance/wasm-micro-runtime*.  Maybe ` if: ${{ github.repository_owner == 'bytecodealliance' }}`

TianlongLiang, 2022-09-29:
> But we need to make sure run the release job only on _bytecodealliance/wasm-micro-runtime_. Maybe ` if: ${{ github.repository_owner == 'bytecodealliance' }}`

Word. I also think about the situation where CI should be triggered on the forked branch, but maybe we could be limited them to less scenery. To think about it, all the commits on the BA branch will be from the pull request, right? Maybe we can only trigger the ci when the branch becomes a pull request base or a pull request target? In that way, CI doesn't have to be run every time sync fork or push a simple commit(if you want to test before PR, you can always trigger it manually). Theoretically, the code quality won't be hurt since we can specify the trigger condition so that CI will run on the base branch when PR is opened/edited and run on the target branch when merged if I understand the [document](https://docs.github.com/en/actions/using-workflows/events-that-trigger-workflows#pull_request_target) correct. I will test whether this logic is feasible if you think this is a good practice

lum1n0us, 2022-09-29:
IIUC, you are saying 
- remove the `on.push` trigger
- only run when a PR is opened, updated and closed with `push_request_target`

```
on:
  pull_request_target:
    types: [opened, synchronize, closed]

jobs:
   xxx:
  if: github.event.pull_request.merged == true
```

Sure, please take a look.

lum1n0us, 2022-10-08:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/1543

closed 2022-10-10 by lum1n0us