wasm-micro-runtime/wasm-micro-runtime #1525: AOT versioning

nchamzn, 2022-09-27:
Hi all,

Is it correct to say that the AOT is only expected to work if the commit hash of the embedded wamr is the same as the commit which was used to build wamrc?

For instance, a module built with a newer version of wamrc might fail to load on old version of wamr like:

```text
09-07 16:05:12.087 10776 10851 I wasm_runtime::: [03:39:42:161 - C7013230]:
09-07 16:05:12.087 10776 10851 I wasm_runtime::: AOT module load failed: resolve symbol aot_check_app_addr_and_convert failed
09-07 16:05:12.087 10776 10851 I wasm_runtime:::
09-07 16:05:12.087 10776 10851 I wasm_runtime::: [03:39:42:281 - C7013230]:
09-07 16:05:12.087 10776 10851 I wasm_runtime::: wasm_module_new failed
09-07 16:05:12.087 10776 10851 I wasm_runtime:::
09-07 16:05:12.087 10776 10851 E WASM    : [WASM:WamrModule_ModuleCreateError] Error creating WAMR module
09-07 16:05:12.087 10776 10851 E WASM    : [WASM:Wamr_ModuleLoadFail] ModuleData failed to load!
```

Are there any plans to support AOT versioning, or would this be too fiddly to implement?

Thanks.

wenyongh, 2022-09-29:
> Is it correct to say that the AOT is only expected to work if the commit hash of the embedded wamr is the same as the commit which was used to build wamrc?
> 
Hi, WAMR AOT runtime tries to support the AOT file generated by older wamrc, but for the AOT file generated by latest wamrc, we had better run it with the latest AOT runtime, as there may be some optimizations applied, which may require new relocation types, etc.

nchamzn, 2022-09-29:
Is there a test I can do to validate that the AOT is compatible?  Is the module successfully loading enough to be confident that it will run without issues?

wenyongh, 2022-09-30:
Had better test some cases to verify the compatibility, for example, for spec cases, you can specify your own wamrc and iwasm:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/tests/wamr-test-suites/spec-test-script/all.py#L25
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/tests/wamr-test-suites/spec-test-script/all.py#L29
and then test then spec cases:
```bash
cd <wamr_root>/tests/wamr-test-suites
./test_wamr.sh -s spec -t aot
```
Also had better test some standalone cases, e.g. `<wamr_root>/tests/benchmarks`.

closed 2022-11-22 by nchamzn