wasm-micro-runtime/wasm-micro-runtime #1528: `__wasi_socket_accept` deceleration conflicted

lum1n0us, 2022-09-28:
The declaration of `__wasi_socket_accept` in [wasi_socket_ext.h](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/libraries/lib-socket/inc/wasi_socket_ext.h#L204) will be conflicted with [api.h](https://github.com/WebAssembly/wasi-libc/blame/05b3b876e63cdc919c79f5127a7ae8dbdb7b095f/libc-bottom-half/headers/public/wasi/api.h#L2015) in *wasi-libc*. 

WASI has supported *soket_accept* since  *wasi-sdk-15*,  and we need to sync up upstream modifications and keep backward compatibility for a while.

P.S.
I can't find any version related macros in [wasi-sdk package](https://github.com/WebAssembly/wasi-sdk/releases/tag/wasi-sdk-16), I guess `#ifdef __WASI_RIGHTS_SOCK_ACCEPT` is one option.

lum1n0us, 2022-09-28:
@loganek please feel free to let me know your opinion.

loganek, 2022-09-28:
Yes unfortunately wasi-libc at the moment is not versioned, so if we want to keep the compatibility with older SDKs, probably the `__WASI_RIGHTS_SOCK_ACCEPT` is the right option.

lum1n0us, 2022-09-28:
@loganek Would you mind arranging the modification?

loganek, 2022-09-28:
Yes, just opened a CR. Please note we're unable to support both new and old definition at the same time in WASM itself - I decided to move to new signature as I think the socket API isn't that widely used yet. `wasi_socket_ext.c` does support both wasi-sdk >= 15 and wasi-sdk < 15, so we just expect any potential users to recompile their applications with no modifications (if they're written in C) or update `__wasi_sock_accept` host call if they use different langauge.

PR #1531 (2022-09-29): Update __wasi_sock_accept signature to match wasi_snapshot_previev1
The function was introduced to WASI about half a year ago after it already existed in WAMR.

It caused problems with compiling `wasi_socket_ext.c` with the wasi-sdk that already had this hostcall exported (wasi-sdk >= 15).

The approach we take is the following:
* we update WASI interface to be compatible with the preview1
* compilation with `wasi_socket_ext.c` supports both wasi_sdk >= 15 and wasi_sdk < 15 (although we intend to drop support for < 15 at one point of time)
* we override `accept()` from wasi-libc - we do that b [... cut]
closed 2022-09-29 by wenyongh