wasm-micro-runtime/wasm-micro-runtime #1533: Failed assertions and return values in `samples/socket-api/wasm-src/socket_opts.c`

lum1n0us, 2022-09-29:
@loganek  we found something in `samples/socket-api/wasm-src/socket_opts.c`:

If run with `./socket_opts`(the native version) and `iwasm ./socket_opts`(the Wasm version), four failures will be raised

Below assertions are failed
- [ ] https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/socket-api/wasm-src/socket_opts.c#L85
- [ ] https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/socket-api/wasm-src/socket_opts.c#L94

Below return values are failed values and `perror()` finds `errno` are both `Invalid argument`
- [ ] https://github.com/bytecodealliance/wasm-micro-runtime/blob/c505da7464b70e009a44c3c7d0a6645f7cc44166/samples/socket-api/wasm-src/socket_opts.c#L251
- [ ] https://github.com/bytecodealliance/wasm-micro-runtime/blob/c505da7464b70e009a44c3c7d0a6645f7cc44166/samples/socket-api/wasm-src/socket_opts.c#L255

cimacmillan, 2022-09-29:
@lum1n0us Thanks for raising, what OS are you using? I can setup the same environment

cimacmillan, 2022-09-30:
I'm going to remove the IPV6_JOIN_GROUP examples from this and add to its own multicast IPV4/6 sample as part of this PR  (revision in progress) https://github.com/bytecodealliance/wasm-micro-runtime/pull/1519

cimacmillan, 2022-10-03:
@lum1n0us I've updated https://github.com/bytecodealliance/wasm-micro-runtime/pull/1519 so that there's a separate multicast ipv4/6 examples which resolves the join/leave group issue. I think the other assertion failures are OS-specific. Eg on linux the send buffer size you request is doubled, but on mac it isn't. 

I'm thinking it might be worth removing the assertions and printing what is received from getting the socket option back, but this example has been useful in testing socket opt changes in WAMR and the assertion easily demonstrates a failure.

closed 2022-10-09 by lum1n0us