wasm-micro-runtime/wasm-micro-runtime #1537: [CI binary release] add a workflow to generate wamr-lldb

lum1n0us, 2022-09-29:
Reference:
https://github.com/xujuntwt95329/lldb_release/blob/main/.github/workflows/release.yml

Plus, it might be a one-time job but it consumes hours. Shall we consider cache the binary content or copy binary files from past releases?

loganek, 2022-10-06:
Our team is looking to improve the debugging experience in WAMR and we can propose a patch for that if you're interested in?

lum1n0us, 2022-10-11:
- [ ] release on MacOS
- [ ] release on Windows ?
- [ ] reuse binaries from previous releases if there is no modification ?
- [ ] or speed up the compilation ?

lum1n0us, 2022-10-12:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/1587

lum1n0us, 2022-10-17:
Using https://github.com/bytecodealliance/wasm-micro-runtime/pull/1605 to speed up.

closed 2022-11-03 by lum1n0us