wasm-micro-runtime/wasm-micro-runtime #1567: [CI binary release] create a workflow for binary release for Windows

lum1n0us, 2022-10-09:
- [ ] build llvm libraries on Windows
- [ ] run a full coverage test of all combination of `WASM_ENABLE_` options
- [ ] build iwasm on Windows
- [ ] build wamrc on Windows
- [ ] run a full features spec test

lum1n0us, 2022-10-09:
Be noticed:
- `WAMR_BUILD_LIBC_UVWASI` should be 1
- Not support `tar` command, just create *.zip* file

tinspin, 2022-11-16:
Hi, how is this progressing?

lum1n0us, 2022-11-30:
We don't have enough resource to start it . It will be very helpful if you are willing to take the task 😃

tinspin, 2022-12-01:
Nobody in their right mind would volontarily compile on Windows.

Until Microsofts allows redistribution of the cl.exe compiler that is!

Oh well, sad that WASM wont happen after all this work...

soyzhc, 2023-02-24:
> Nobody in their right mind would volontarily compile on Windows.
> 
> Until Microsofts allows redistribution of the cl.exe compiler that is!
> 
> Oh well, sad that WASM wont happen after all this work...

why not use mingw？

tinspin, 2023-02-24:
You want the compiler to be portable in a zip and I have that for cl.exe now after extracting it from VS.

You might be able to use this in the same way but I haven't tried because I have my setup working:

https://github.com/mstorsjo/llvm-mingw/releases

CrosRoad95, 2023-02-24:
can anyone provide step by step instructions how can i do this? :)

tinspin, 2023-02-24:
I think if they had that they would have compiled it already.

tinspin, 2024-12-21:
Great to see that you finally made a iwasm.exe, but you probably should compile it to 32-bit too because around half of the projects on Windows are still running 32-bit since there is no reason to move to 64-bit there until Intel removes 32-bit from their CPUs and even then they will be properly emulated I'm sure.

Also would be great with a way to use iwasm from your project and for that you need a .dll and a simple tutorial, sort of what was done here:

https://github.com/bytecodealliance/wasm-micro-runtime/issues/85