wasm-micro-runtime/wasm-micro-runtime #1571: [CI Optimization] Add Fast-JIT into CIs

lum1n0us, 2022-10-10:
- [ ] Add `WAMR_BUILD_FAST_JIT` into iwasm compilation combination test
- [ ] Add `WAMR_BULID_FAST_JIT` into wasm-c-api compilation combination test
- [ ] Add *fast-jit* into spec test jobs
- [ ] Adjust spec test feature combination
- [ ] Speed up spec test by downloading the released WABT binary instead of compiling from source code
- [ ] Remove `MC_JIT_BUILD_OPTIONS`?

lum1n0us, 2022-10-10:
| OS                 | defulat               | +multi module | +simd      | +thread              |
| ------------------ | --------------------- | ------------- | ---------- | -------------------- |
| ubuntu-2004        | INTERP(2)+JIT(LF)+AOT | INTERP(2)     | JIT(L)+AOT | INTERP(2)+JIT(L)+AOT |
| ubuntu-2004 x86_32 | INTERP(2)+AOT         | NA            | NA         | INTERP(2)+AOT        |
| ubuntu-2004 sgx    | INTERP(2)+AOT         | NA            | AOT        | NA                   |

> - JIT(LF): LLVM JIT and FAST JIT
> - JIT(L): LLVM JIT
> - JIT(F): FAST JIT
> - SGX enables `thread` option by default

wenyongh, 2023-04-27:
Close as all the tasks were done

closed 2023-04-27 by wenyongh