wasm-micro-runtime/wasm-micro-runtime #1582: aot not working on armv7a

morn-0, 2022-10-12:
[image]

hangedfish, 2022-10-12:
try "armv7"

https://github.com/bytecodealliance/wasm-micro-runtime/blob/f28a15bf2f6ded8ed2e6b4df41d132977664b80b/core/iwasm/compilation/aot_llvm.c#L1126

morn-0, 2022-10-12:
> try "armv7"
> 
> https://github.com/bytecodealliance/wasm-micro-runtime/blob/f28a15bf2f6ded8ed2e6b4df41d132977664b80b/core/iwasm/compilation/aot_llvm.c#L1126

armv7 includes armv7a is it

morn-0, 2022-10-12:
but segmentation fault

nchamzn, 2022-10-12:
> but segmentation fault

Any further information? stack trace or signal?

shengyun-zhou, 2022-10-13:
I encountered this issues too, here is my current solution:
1. Use `armv6` target. `armv7` may crash at runtime by SIGBUS(Invalid address alignment).
2. Enable hard-float explicitly by adding command line option `--cpu-features=+vfp2`.
3. Target ABI must be set to either `--target-abi=eabi` or `--target-abi=gnueabihf` for linux platform. Incorret target ABI will cause segment fault. This value depends on which ABI your cross compiler use to build `iwasm`. You can check if macro `__ARM_PCS_VFP` is defined by your cross compiler to determine if `--target-abi=gnueabihf` should be used, eg.

```shell
armv7a-linux-gnueabihf-gcc -dM -E - < /dev/null | grep __ARM_PCS_VFP
```

So, the final `wamrc` command line should look like:

```shell
wamrc --target=armv6 --target-abi=gnueabihf --cpu=generic --cpu-features=+vfp2 -o ./test.aot test.wasm
```

nchamzn, 2022-10-13:
> Use armv6 target. armv7 may crash at runtime by SIGBUS(Invalid address alignment).

We also found this but after some debugging noticed it was due to passing unaligned address to some NEON instructions, we worked around for now by disabling NEON: ``--cpu-features=-neon``.

morn-0, 2022-10-18:
It seems that the above methods did not work for me. I am android armv7a and I have observed that sigv_maperr will appear when creating an instance.

morn-0, 2022-10-18:
> It seems that the above methods did not work for me. I am android armv7a and I have observed that sigv_maperr will appear when creating an instance.

I am an arm64 android device, but he is compatible with armv7a. i don't know if this will have any effect

hangedfish, 2022-10-18:
> > It seems that the above methods did not work for me. I am android armv7a and I have observed that sigv_maperr will appear when creating an instance.
> 
> I am an arm64 android device, but he is compatible with armv7a. i don't know if this will have any effect

Try add "--target-abi=eabi". I am sure I have run successfully in Android armeabi-v7a.

closed 2023-04-23 by wenyongh