wasm-micro-runtime/wasm-micro-runtime #1629: Unable to build the sample socket build api

jakob999999, 2022-10-21:
I ran into the following issues when I try to build the sample Socket API. Has anyone had the same issues, because it seems like there is something wrong with the declarations of the files provided?

[image]

loganek, 2022-10-22:
Hi @jakob999999 , we recently migrated to newer wasi-sdk, and the build most likely fails because you still use older version. Can you please try to install wasi-sdk version 15 or newer?

See https://github.com/bytecodealliance/wasm-micro-runtime/issues/1528 for details.

mmathys, 2022-10-26:
Had the same issue, fixed it updating wasi-sdk

closed 2023-04-27 by wenyongh