wasm-micro-runtime/wasm-micro-runtime #1644: Disable Shared Memory CMake flag not working

mmathys, 2022-10-26:
I am trying to compile `tcp-server.wasm` in the [socket-api sample](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/socket-api), but without `pthread` support and without shared memory. I modified added these configs into `CMakeLists.txt` to disable `pthread` and disable shared memory:

```cmake
set(WAMR_BUILD_LIB_PTHREAD 0)
set(WAMR_BUILD_LIB_PTHREAD_SEMAPHORE 0)
set(WAMR_BUILD_SHARED_MEMORY 0)
```

and remove the pthread library from the `tcp-server` executable:

```cmake
add_executable(tcp_server ${CMAKE_CURRENT_SOURCE_DIR}/wasm-src/tcp_server.c)
# target_link_libraries(tcp_server pthread)
```

and modified that `wasm-src/tcp-server.c` does not use `pthread`. However, the compiled WebAssembly module still shows shared memory:

```
$ wasm-objdump tcp_server.wasm -x

...

Memory[1]:
 - memory[0] pages: initial=2 max=3 shared

...
```

How can I completely disable shared memory?

xujuntwt95329, 2022-11-02:
Hi @mmathys 

`WAMR_BUILD_xxxx` is used to control the compilation of runtime, but what you want to do is to control the compilation of wasm.

You should remove these lines to disable the shared memory feature and pthread in generated wasm module
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/socket-api/wasm-src/CMakeLists.txt#L52
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/socket-api/wasm-src/CMakeLists.txt#L57

mmathys, 2022-11-02:
Thanks a lot, @xujuntwt95329!

closed 2023-04-23 by wenyongh