wasm-micro-runtime/wasm-micro-runtime #1650: fatal error: autoconf.h: No such file or directory

goodluckrhj, 2022-10-27:
I followed the steps to build the WAMR SDK for a configuration profile.

Firstly, i download binary and installed(extracted) WASI SDK to working dir instead of "/opt"

Then, i modified wamr_config_default.cmake as the following

```
set (WAMR_BUILD_TARGET "XTENSA")
set (WAMR_BUILD_PLATFORM "zephyr")
set (WAMR_BUILD_AOT 1)
set (WAMR_BUILD_INTERP 0)
set (WAMR_BUILD_LIBC_BUILTIN 1)
set (WAMR_BUILD_APP_LIST WAMR_APP_BUILD_ALL)
set (WAMR_BUILD_APP_FRAMEWORK 1)

```
Finally, i build_sdk.sh -n esp32-zephyr-wamr with the modified setting.

It seemed successfully built app-sdk under the "out" dir. 
However, failed to build runtime-sdk, with the falling output:
```

##############  Start to build runtime sdk  ###############
CMake Deprecation Warning at CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 11.3.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Add native module base
Add native module connection
Add native module sensor
Add native module wgl
-- Performing Test M32_OK
-- Performing Test M32_OK - Failed
-- Build Configurations:
     Build as target XTENSA
     CMAKE_BUILD_TYPE Release
     WAMR Interpreter disabled
     WAMR AOT enabled
     WAMR Fast JIT disabled
     WAMR LLVM ORC JIT disabled
     Libc builtin enabled
     Libc WASI disabled
     Fast interpreter disabled
     Multiple modules disabled
     Reference types disabled
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tsinghua/Projects/wamr_space/wasm-micro-runtime/wamr-sdk/runtime/build_runtime_sdk
Scanning dependencies of target vmlib
[  0%] Building C object CMakeFiles/vmlib.dir/home/tsinghua/Projects/wamr_space/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c.o
In file included from /home/tsinghua/Projects/wamr_space/wasm-micro-runtime/core/shared/platform/zephyr/../include/platform_common.h:13,
                 from /home/tsinghua/Projects/wamr_space/wasm-micro-runtime/core/shared/platform/zephyr/../include/platform_api_vmcore.h:9,
                 from /home/tsinghua/Projects/wamr_space/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c:6:
/home/tsinghua/Projects/wamr_space/wasm-micro-runtime/core/shared/platform/zephyr/platform_internal.h:9:10: fatal error: autoconf.h: No such file or directory
    9 | #include <autoconf.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/vmlib.dir/build.make:76: CMakeFiles/vmlib.dir/home/tsinghua/Projects/wamr_space/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/vmlib.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
Failed to build runtime sdk
```

How can i fix the error of missing "autoconf.h"?

wenyongh, 2022-11-02:
Hi, did you set the zephyr environment firstly? e.g. `source <zephyr_dir>/zephyr-env.sh` and then try again?

goodluckrhj, 2022-11-04:
> Hi, did you set the zephyr environment firstly? e.g. `source <zephyr_dir>/zephyr-env.sh` and then try again?

Hi, Wenyong, thanks for your reply. I fixed this problem as you suggested to check the environment.

goodluckrhj, 2022-11-16:
Hi, @wenyongh 
I decided to implement wamr on the platform of "zephyr+esp32", and the dev-environment has changed to

- Ubuntu 22.04
- zephyr 3.2.99 latest version 
- esp-idf 4.43

While i have succeeded build applications for "zephyr+esp32", when it comes to build WAMR-SDK, a same "runtime-app, no autoconf.h"  error appears again.

goodluckrhj, 2022-11-21:
I have solved the problems:
https://github.com/bytecodealliance/wasm-micro-runtime/issues/1726#issuecomment-1321353237

closed 2023-04-23 by wenyongh