wasm-micro-runtime/wasm-micro-runtime #1659: How to call wasm function pointer and parse va_list in native api ?

hangedfish, 2022-10-30:
Hi, I have a native api that needs a callback function and va_list but I don't know how to implement it.

Please give me some help.

Below is a simplified test case:

WASM Source:

```C++
// wasm source

// imported from host
export "C"
void call_int_method(void* func, va_list args);

int add(int a, int b) {
    return a+b;
}

int main(int args, char*[] argv) {
    int sum = call_int_method(add, 1, 2);
    printf("%d\n", sum);
    return 0;
}
```

Host Native Function:
```C++
// Host Native Function

// {"call_int_method", "(**)i"}
int wasm_call_int_method_wrapper(wasm_exec_env_t exec_env, uint32_t func_ptr_offset, _va_list va_args) {
   // How to call func_ptr ?
   // How to parse va_list ?
}

```

xujuntwt95329, 2022-11-02:
Hi, 

- **For function pointer:**

    The actual value you get in native (`func_ptr_offset` in your example) is the `table element index` of the function pointer, and then you can use `wasm_runtime_call_indirect` API to call this function through its element index

    https://github.com/bytecodealliance/wasm-micro-runtime/blob/9ba8c8957d114e2c742322369c04cb74cf2cf0c1/core/iwasm/common/wasm_runtime_common.h#L595-L597

    We didn't expose this API in `wasm_export.h` you can declare it in your source code.

- **For va_args:**
    you can refer to the `printf_wrapper` function,
    https://github.com/bytecodealliance/wasm-micro-runtime/blob/9ba8c8957d114e2c742322369c04cb74cf2cf0c1/core/iwasm/libraries/libc-builtin/libc_builtin_wrapper.c#L398-L412

    we defined `_va_list` as `char *`, and use `_va_arg` to get a value and move the pointer. You can use similar approach in your code
    https://github.com/bytecodealliance/wasm-micro-runtime/blob/9ba8c8957d114e2c742322369c04cb74cf2cf0c1/core/iwasm/libraries/libc-builtin/libc_builtin_wrapper.c#L53

    Note that the va_args must be carefully checked to avoid out of bound memory access

hangedfish, 2022-11-02:
@xujuntwt95329 
Thanks for your reply, I got it. :)

closed 2022-11-02 by hangedfish