wasm-micro-runtime/wasm-micro-runtime #1661: Support for FreeBSD?

girgen, 2022-10-31:
Could we get WAMR to provide cmake for freebsd. I browsed the source and go as far as that it is just not implemented yet. The interface needs to be implemented for that platform, AFAICS. Copying the Darwin config might be a good start?

See https://github.com/fluent/fluent-bit/issues/6289#issue-1425488799 for details. 

Thanks!
Palle

TianlongLiang, 2022-11-02:
Thanks for bringing this issue to our attention, I will start looking into it. I will keep you posted on the progress

TianlongLiang, 2022-11-14:
I finished setting up my FreeBSD environment(FreeBSD newbie here, it took me some time:joy:), and I plan to try [port tutorial](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/port_wamr.md). Hopefully, I can make it works in the next few days. :smile:

TianlongLiang, 2022-11-21:
I follow the tutorial I mentioned, get it able to compile on FreeBSD, and push the code [to my fork](https://github.com/TianlongLiang/wasm-micro-runtime/tree/dev/freebsd). I will start some further testing in the next few days. Meanwhile, I would appreciate it if you could try it too and tell me any problem you encounter. @girgen :laughing:

TianlongLiang, 2022-12-28:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/1825 pr for FreeBSD support was merged, any more suggestions or questions?

girgen, 2023-02-07:
Looks great, AFAICT. Thanks!

closed 2023-02-07 by girgen