wasm-micro-runtime/wasm-micro-runtime #1665: Question: Why is pthread required when -DWAMR_BUILD_LIB_PTHREAD=0

phated, 2022-10-31:
I'm building iwasm core with `-DWAMR_BUILD_LIB_PTHREAD=0` but when I use the library in my bindings, the build fails because pthread is missing. I can link in pthread and it works, but why is pthread required when I built **without** pthread support?

wenyongh, 2022-11-02:
Hi, there are some reasons: (1) developer may run different module instances in individual threads created by him (but not runtime), and we need to allocate memory for these module instances, if we use a global heap buffer managed by runtime, we need to add lock for it, (2) the native thread's stack overflow check is enabled by runtime, normally we need to get its stack min address (boundary), (3) sometimes we may hope to get current thread info, e.g. native thread handle

phated, 2022-11-03:
Thank you for the explanation! I opened #1681 as a solution for providing these type of requirements using pkg-config.

closed 2022-11-03 by phated