wasm-micro-runtime/wasm-micro-runtime #1670: sgx-ra returns 0xe019

JamesMenetrey, 2022-11-01:
Dear WAMR developers,

I'm trying to run the sample `sgx-ra`, but so far, I've got the following errors:

```
[get_platform_quote_cert_data ../qe_logic.cpp:378] Error returned from the p_sgx_get_quote_config API. 0xe019
[ERROR] rats_ocall_get_target_info()@L31: sgx_qe_get_target_info() with error code 0xe019
[get_platform_quote_cert_data ../qe_logic.cpp:378] Error returned from the p_sgx_get_quote_config API. 0xe019
[ERROR] rats_ocall_qe_get_quote_size()@L38: sgx_qe_get_quote_size(): 0xe019
[ERROR] sgx_ecdsa_collect_evidence()@L116: sgx_qe_get_quote_size(): 0x0000, 0xa000e019
[ERROR] librats_collect_evidence_to_json()@L114: failed to librats_collect_evidence return 0xa000e019
Librats collect evidence failed.
```

Regarding my configuration, I have successfully completed the installation of all the components, as described in the README. 

Googling around the error code "0xe019", I discover this may be due to the missing component PCCS. So, I have installed it following [this guide](https://www.intel.com/content/www/us/en/developer/articles/guide/intel-software-guard-extensions-data-center-attestation-primitives-quick-install-guide.html), subscribed to the PCS API from Intel, verified that my chip could be verified using the tool `PCKIDRetrievalTool`, but I still get the same error from the WAMR's sample `sgx-ra`.

Do you have an idea why I get this error?
Does this sample require the installation of PCCS? If so, I think we may better document the installation part in the README.

For completeness, here is the output of `sgx-detect`:
```
❯ sudo ./sgx-detect
Detecting SGX, this may take a minute...
✔  SGX instruction set
  ✔  CPU support
  ✔  CPU configuration
  ✔  Enclave attributes
  ✔  Enclave Page Cache
  SGX features
    ✔  SGX2  ✔  EXINFO  ✘  ENCLV  ✘  OVERSUB  ✘  KSS
    Total EPC size: 94.0MiB
✔  Flexible launch control
  ✔  CPU support
  ✔  CPU configuration
  ✔  Able to launch production mode enclave
✔  SGX system software
  ✔  SGX kernel device (/dev/sgx_enclave)
  ✔  libsgx_enclave_common
  ✔  AESM service
  ✔  Able to launch enclaves
    ✔  Debug mode
    ✔  Production mode
    ✔  Production mode (Intel whitelisted)

You're all set to start running SGX programs!
```

Cheers,
Jämes

wenyongh, 2022-11-02:
@zeuson0 Could you help check the issue? Since you are the author and we are not familiar with it. Thanks a lot.

zeuson0, 2022-11-03:
Hi, thanks for reminding us of the missing description of PCCS in the README, we will fix it soon. Error code "0xe019" means that it failed to fetch the collateral from the PCCS because of the "NETWORK_ERROR". Mostly, this is because the PCCS is not installed successfully or the configuration of PCCS is not initialized correctly. 

You could check the installation through the command "curl -k https://localhost:8081/sgx/certification/v3/qe/identity". If failed, you'd better check what went wrong during the installation of PCCS and reinstall it.

You could check the configuration in `/etc/sgx_default_qcnl.conf` to make sure the `pccs_url` is set correctly and `use_secure_cert` is set to be `false`.

If neither is a problem, you cloud provide the logs of the PCCS installation, as well as the output of PCKIDRetrievalTool and `/etc/sgx_default_qcnl.conf`, and I will help you to check it out.

JamesMenetrey, 2022-11-03:
Hey @zeuson0,

Many thanks for your response! The configuration of `use_secure_cert` to `false` in `/etc/sgx_default_qcnl.conf` was the missing bit. I now get the `Evidence is trusted`.

I can dedicate some time to updating the documentation for WAMR for the installation of the PCCS. I will reproduce the setup on another machine and confirm that it works as expected.

Lastly, may I ask you a favour and check my question in the Discussions tab, please: #1664 ? It is directly connected to the RA feature, and I think you can have some interesting insights to share. :)

Many thanks!

closed 2022-11-03 by JamesMenetrey