wasm-micro-runtime/wasm-micro-runtime #1681: Generate pkg-config for the project

phated, 2022-11-03:
While trying to write bindings for this project, I've discovered that WAMR needs various flags. For example, it always needs [-lpthread](https://github.com/bytecodealliance/wasm-micro-runtime/issues/1665) or `-liphlpapi -luserenv -luv_a` when building with `WAMR_BUILD_LIBC_UVWASI`.

I noticed that `libuv` already provides a list of these flags for itself [via pkg-config](https://github.com/libuv/libuv/blob/v1.44.2/libuv-static.pc.in) and I was hoping that WAMR could provide a complete list of these flags based on build configuration and platform.

lum1n0us, 2022-11-04:
Thanks. It is a good suggestion. We will take a look and see what we can do.

haasn, 2022-11-08:
+1 on this. Supplying a proper `.pc` file is a near-requirement to be able to use WAMR effectively from within C/C++ projects on Linux.

lum1n0us, 2022-12-04:
I am trying to imagine "how to properly integrate WAMR into a project which is not use cmake". Does it like?
- Download WAMR source code and compile
- Install. Maybe manually
- Link WAMR with the target project

behdad, 2023-02-25:
I've also stumbled on this. Specially if WAMR is built with llvm I need to pass -lLLVM-15 currently. I don't know how to link to the included LLVM, since that doesn't seem to be installed by cmake. If I can be advised of that I can give it a try to add a pkg-config file.