wasm-micro-runtime/wasm-micro-runtime #1683: An error was generated when running the function "wasm_runtime_instantiate".  error_buf: uvwasi init failed

kamylee, 2022-11-04:
I attempt embed wamr into my c++ source code.
I added some settings to "CmakeLists.txt" file.

```
	set (WAMR_BUILD_PLATFORM "windows")
	# Build as X86_64 by default in 64-bit platform
    set (WAMR_BUILD_TARGET "X86_64")
	set (WAMR_BUILD_INTERP 1)
	set (WAMR_BUILD_FAST_INTERP 1)
	set (WAMR_BUILD_AOT 1)
	set (WAMR_BUILD_LIBC_BUILTIN 1)
	set (WAMR_BUILD_LIBC_WASI 1)
	set (WAMR_BUILD_SIMD 1)
	if (NOT DEFINED WAMR_BUILD_LIBC_UVWASI)
	  # Enable libc uvwasi support by default
	  set (WAMR_BUILD_LIBC_UVWASI 0)
	endif ()
	set(WAMR_ROOT_DIR "D:/WorkSpace/wasm/wasm-micro-runtime")
	include(${WAMR_ROOT_DIR}/build-scripts/runtime_lib.cmake) 
...
TARGET_LINK_LIBRARIES(zm... libcef libcef_dll_wrapper  vmlib libiwasm)
```

When it runs to the  wasm_runtime_instantiate  function, an error will be generated: uvwasi init failed
```
if (!(wasm_module_inst =
		wasm_runtime_instantiate(wasm_module, stack_size, heap_size,
			error_buf, sizeof(error_buf)))) {
		printf("%s\n", error_buf);
		return;
	}
//error_buf's value is uvwasi init failed .
```
What cause this? How can I fix him?

TianlongLiang, 2022-11-04:
Hi, there. You could first try to call [function `wasm_runtime_set_wasi_args()`](https://github.com/bytecodealliance/wasm-micro-runtime/blob/aba267afc79e1154cac13701d7f7c33a970331df/product-mini/platforms/posix/main.c#L613-L614) after calling `wasm_runtime_load()`.

If that doesn't work, could you please try to remove `WAMR_BUILD_LIBC_WASI` and `WAMR_BUILD_LIBC_UVWASI` in CMake and see if the error persists?

kamylee, 2022-11-04:
I've modified it like this.the error still exists.
removed  WAMR_BUILD_LIBC_WASI and WAMR_BUILD_LIBC_UVWASI.
```
char error_buf[128] = { 0 };
	wasm_module_t wasm_module = NULL;
	wasm_module_inst_t wasm_module_inst = NULL;
	wasm_function_inst_t func;
	wasm_exec_env_t exec_env;
	uint32 stack_size = 16 * 1024, heap_size = 16 * 1024;

/* initialize the wasm runtime by default configurations */
wasm_runtime_init();

/* read WASM file into a memory buffer */
char wasm_path[255] = {0};
	strcpy_s(wasm_path, "html\\hello.wasm");
	
	uint32 wasm_file_size;
	uint8_t* wasm_file_buf = NULL;
	wasm_file_buf = (uint8_t*)bh_read_file_to_buffer(wasm_path, &wasm_file_size);
	if (!wasm_file_buf) {
		printf("Open wasm app file [%s] failed.\n", wasm_path);
		return;
	}

/* load WASM module */
	if (!(wasm_module = wasm_runtime_load(wasm_file_buf, wasm_file_size,
		error_buf, sizeof(error_buf)))) {
		printf("%s\n", error_buf);
		return;
	}

	const char* dir_list[8] = { NULL };
	uint32 dir_list_size = 0;
	const char* env_list[8] = { NULL };
	uint32 env_list_size = 0;

	wasm_runtime_set_wasi_args(wasm_module, dir_list, dir_list_size, NULL, 0,
		env_list, env_list_size, argv, argc);

	/* instantiate the module */
	if (!(wasm_module_inst =
		wasm_runtime_instantiate(wasm_module, stack_size, heap_size,
			error_buf, sizeof(error_buf)))) {
		printf("%s\n", error_buf);
		return;
	}

	exec_env = wasm_runtime_create_exec_env(wasm_module_inst, stack_size);
	if (!exec_env) {
		printf("Create wasm execution environment failed.\n");
		return;
	}

	if (!(func = wasm_runtime_lookup_function(wasm_module_inst, "generate_float",
		NULL))) {
		printf("The generate_float wasm function is not found.\n");
		return;
	}

```

kamylee, 2022-11-04:
I used this code:
#Set_Target_Properties(... PROPERTIES LINK_FLAGS_RELEASE "/SUBSYSTEM:WINDOWS /ENTRY:mainCRTStartup")
comment it and it's work properly!

TianlongLiang, 2022-11-04:
That's great! I'm glad it's all worked out.

closed 2023-04-23 by wenyongh