wasm-micro-runtime/wasm-micro-runtime #1688: Configure linter for VSCode extension code

loganek, 2022-11-07:
While reviewing the PR https://github.com/bytecodealliance/wasm-micro-runtime/pull/1637/ I noticed there's a bit of inconsistency in our code for VS Code extension. I'm opening this issue to track setting up a linter that:
1. checks for style
2. detects common errors
3. drives overall best practices in the code

I'd like to also use this issue to track the fixes.

lum1n0us, 2022-11-10:
May I ask 
- Will it be a linter for js/ts only?
- Will it be an offline linter(running on developer's computer) or online CI linter?

loganek, 2022-11-15:
Hi @lum1n0us I think we can start with ts/js linter for now and only for the VSCode extension. Ideally the linter should run in the CI, but also we should let developers to run them locally with autofix (this could be e.g. a `lint` script in `package.json` so developers can simply execute it with `npm run lint`).

NingW101, 2022-12-16:
@loganek Thanks for your helpful advice 👍  
`ESlint` has been set up for `vscode-extesnion` codebase and I filed #1817 to fix the reported warnings and errors including renaming following the convention and some other changes. PTAL~

loganek, 2022-12-21:
Many thanks for doing that!

closed 2023-04-27 by wenyongh