wasm-micro-runtime/wasm-micro-runtime #1690: Malloc returns zero when exported for larger AOT binaries

robinvanemden, 2022-11-08:
We autogenerate AI models and make them available as AOT binaries that can be run on different architectures using WAMR. Some time ago, we changed from non-exported to exported `malloc` and `free` on compiling our binaries, using:

``   -Wl,--export=__heap_base -Wl,--export=__data_end -Wl,--export=malloc -Wl,--export=free \``

But we now encounter a problem with larger (1gb+ of mem) AI models. For these larger models, the within-wasm `malloc` returns zero on allocating additional memory. We do not encounter this issue when not exporting malloc and free. Also, smaller models (for instance, 500MB of memory all in all) run fine for both exported and non-exported malloc. Do you have a suggestion on what might be happening here?

robinvanemden, 2022-11-09:
We did some more tests, trying different config options. Among others, we thought we'd try to remove dlmalloc from the WASI libc, to see if that might somehow interfere with our setup:

```
/opt/wasi-sdk/bin/llvm-ar -d /opt/wasi-sdk/share/wasi-sysroot/lib/wasm32-wasi/libc.a dlmalloc.o
```
Then, we recompiled our wasi binary with `malloc` export:

```
-Wl,--export=malloc -Wl,--export=free -Wl,--export=__heap_base,--export=__data_end \
-Wl,--export=malloc_buffer,--export=free_buffer,--export=get_out_len,--export=get_out_loc \
```

On doing that, we got `failed to link import function ()` for malloc, realloc and free. So we thought we might provide it, by changing the following settings on compiling our main application:

``` 
SET(WAMR_BUILD_LIBC_BUILTIN 0)
SET(WAMR_BUILD_LIBC_WASI 1)
``` 

.. which worked fine when not exporting free/malloc, to:

```
SET(WAMR_BUILD_LIBC_BUILTIN 1)
SET(WAMR_BUILD_LIBC_WASI 1)
```

Now, everything is working again. But which malloc/free config is the the most optimal / safe? And why does "just" exporting malloc and free without removal of dlmalloc work well, up to a memory limit?

wenyongh, 2022-11-14:
@robinvanemden The wasm app malloc's (or malloc_buffer's) behavior and runtime api wasm_runtime_module_malloc's behavior depend on how the app heap and libc heap are configured, or, whether the memory is allocated from the app heap or the libc's heap, refer to the WAMR memory model for more details:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/memory_tune.md#the-memory-model

(1) If the libc's heap management library (like dlmalloc library) is compiled into wasm bytecode, then the wasm app's malloc function will apply memory from it. In this case, the linear memory is growable, when there is no heap space left to allocate, the libc's heap management will grow the whole linear memory firstly and then allocate a new memory chunk for wasm app. As growing the whole linear memory may require the runtime to re-allocate a relatively larger memory, runtime may fail to do that if the required linear memory size is larger than the maximum linear memory size specified, or there is no space left in the global heap (seems that global heap space is enough for you as you set to use Alloc_With_System_Allocator). So could you please enlarge the maximum linear memory size while building wasm app:
```
For wasi-sdk, the initial/max size can be specified with -Wl,--initial-memory=n1,--max-memory=n2,
for emsdk, the initial/max size can be specified with -s INITIAL_MEMORY=n1 -s MAXIMUM_MEMORY=n2 -s ALLOW_MEMORY_GROWTH=1 or -s TOTAL_MEMORY=n,
and for asc, they can be specified with --initialMemory and --maximumMemory flags.
```
And in this case, if the malloc/free function is exported, then the app heap won't be created by runtime, and wasm_runtime_module_malloc will invoke the exported malloc function to allocate memory. And if the malloc/free function isn't exported, the app heap is created and wasm_runtime_module_malloc applies memory from it.

(2) If the libc's heap management library (like dlmalloc library) isn't compiled into wasm bytecode, e.g. remove dlmalloc.o from libc.a, then both wasm app's malloc function and runtime api wasm_runtime_module_malloc allocate memory from the app heap created by runtime. In this case, the app heap size is fixed and specified by `wasm_runtime_instantiate`, and the linear memory isn't growable. The libc-builtin should be compiled and linked, so wasm app can call the imported malloc function.

We think both methods work, it is up to you. And had better just choose one to reduce the footprint: use libc's heap, export  malloc/free and configure a larger linear maximum memory, or use app heap with a larger size and remove dlmalloc.o from libc.a. Note that the latter is thread-safe as runtime adds lock for each alloc/free operation, and the former isn't thread-safe by default and should not be used when multi-thread feature is required.

wenyongh, 2022-11-18:
@robinvanemden Maybe you can debug the source code of wasi-libc and refine its memory allocation algorithm:
https://github.com/WebAssembly/wasi-libc/blob/30094b6ed05f19cee102115215863d185f2db4f0/dlmalloc/src/malloc.c

You can refer to the document of wasi-sdk to build it from source code:
https://github.com/WebAssembly/wasi-sdk

Or refer to WAMR's build-wasi-sdk script to build it:
https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/test-tools/build-wasi-sdk

robinvanemden, 2022-11-26:
Thanks, @wenyongh. We will follow your suggestions and see whether we are able to get to the bottom of this issue and report back our findings.