wasm-micro-runtime/wasm-micro-runtime #1693: Cannot call multiple functions consecutively through API in "wasm_export.h"

YushuoEdge, 2022-11-10:
Hi, I want to use the Function APIs (in wasm_export.h) to continuously call multiple Functions in one module instance in a program. 
**The module bytecode is:**
```
  (func $get_params_buffer_size (type $t16) (result i32)
    i32.const 16384)
  (func $get_results_buffer_size (type $t16) (result i32)
    i32.const 4096)
  (func $get_params_buffer_addr (type $t16) (result i32)
    i32.const 18448)
  (func $get_results_buffer_addr (type $t16) (result i32)
    i32.const 34832)
```

**The c++ code is as below:**
```
    wasm_function_inst_t func1 = wasm_runtime_lookup_function(module_inst, "get_params_buffer_size", "()i");
    if (wasm_runtime_call_wasm_a(exec_env, func1, 1, params_buffer_size, 0, nullptr)) {
        printf("Successful Execution func1!\n");
    } else {
        printf("Execute function func1 failed. Error: %s\n", wasm_runtime_get_exception(module_inst));
    }

    wasm_function_inst_t func2 = wasm_runtime_lookup_function(module_inst, "get_results_buffer_size", "()i");
    if (wasm_runtime_call_wasm_a(exec_env, func2, 1, results_buffer_size, 0, nullptr)) {
        printf("Successful Execution func2!\n");
    } else {
        printf("Execute function func2 failed. Error: %s\n", wasm_runtime_get_exception(module_inst));
    }

    wasm_function_inst_t func3 = wasm_runtime_lookup_function(module_inst, "get_params_buffer_addr", "()i");
    if (wasm_runtime_call_wasm_a(exec_env, func3, 1, params_buffer, 0, nullptr)) {
        printf("Successful Execution func3!\n");
    } else {
        printf("Execute function func3 failed. Error: %s\n", wasm_runtime_get_exception(module_inst));
    }

    wasm_function_inst_t func4 = wasm_runtime_lookup_function(module_inst, "get_results_buffer_addr", "()i");
    if (wasm_runtime_call_wasm_a(exec_env, func4, 1, results_buffer, 0, nullptr)) {
        printf("Successful Execution func4!\n");
    } else {
        printf("Execute function func4 failed. Error: %s\n", wasm_runtime_get_exception(module_inst));
    }
```

But I can always only successfully call the first function. The result of running the program is:
```
Successful Execution func1!
Execute function func2 failed. Error: Exception: uninitialized element
Execute function func3 failed. Error: Exception: uninitialized element
Execute function func4 failed. Error: Exception: uninitialized element
params buffer size: 16384 
params buffer addr: 0 
results buffer size: 0 
results buffer addr: 0 
```

The `version` I used is `WAMR-1.1.0`. I have set:
```
set (WAMR_BUILD_INTERP 1)
set (WAMR_BUILD_FAST_INTERP 1)
set (WAMR_BUILD_AOT 1)
set (WAMR_BUILD_LIBC_WASI 1)
set (WAMR_BUILD_BULK_MEMORY 1) 
```

What I want to ask is: can I use it like this? Or do you need to open some configuration when using it like this？

YushuoEdge, 2022-11-10:
I seem to have found the problem. When I include iostream in the module source code, there will be a problem, but if I replace it with stdio.h, it will run smoothly. The problem lies in the WASI-SDK?

xujuntwt95329, 2022-11-10:
Could you re-build the wasm module with `-mexec-model=reactor` option? 

Seems it's related to the command/reactor mode, see https://github.com/WebAssembly/wasi-libc/pull/264

YushuoEdge, 2022-11-11:
> Could you re-build the wasm module with `-mexec-model=reactor` option?
> 
> Seems it's related to the command/reactor mode, see [WebAssembly/wasi-libc#264](https://github.com/WebAssembly/wasi-libc/pull/264)

It works for me. Thanks!

closed 2022-11-11 by YushuoEdge