wasm-micro-runtime/wasm-micro-runtime #1694: Advice for reducing rust WASM app footprint

Therdel, 2022-11-10:
[Build WASM applications](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md) has a paragraph "[How to reduce the footprint?](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md#2-how-to-reduce-the-footprint)" for building C code with the WASI-SDK.

The Rust part could use equivalent suggestions.
One of which is to reduce the (auxiliary?)stack size down from the default 1MB(!) - which is too much for many MCU scenarios.
Passing linker arguments is possible in a `[workspace root]/.cargo/config.toml` file:
```toml
rustflags = [
  "-C", "link-arg=-zstack-size=8192",
]
```
Another one: Not using the print-formatting infrastructure reduced my binary size from 22680 to 237 Bytes.

wenyongh, 2022-11-14:
@Therdel Many thanks for the advice! We will update the document according to you suggestion.

Therdel, 2022-11-14:
You're welcome!

wenyongh, 2022-11-15:
@Therdel I updated the document in #1701, thanks.

closed 2022-11-25 by wenyongh