wasm-micro-runtime/wasm-micro-runtime #1708: WASM_OP_SELECT does not handle 64 bit operands

dbaum, 2022-11-15:
I believe WASM_OP_SELECT (opcode 0x1b) is supposed to handle both 32-bit and 64-bit operands.  At least that's my interpretation of https://webassembly.github.io/multi-value/core/syntax/instructions.html stating "Instructions in this group can operate on operands of any value type.".  For example, emscripten will compile

```
int64 Max(int64 x, int64 y) {
  return x > y ? x : y;
}
```

into 

```
(func (;1;) (type 1) (param i64 i64) (result i64)
    local.get 0
    local.get 1
    local.get 0
    local.get 1
    i64.gt_s
    select)
```
Note that the final opcode will use a 32-bit stack value to select between two other 64-bit stack values.

This generated code appears to work in the interpreter but fail JIT/AOT compilation.  I believe it is because core/wasm/copmilation/aot_compiler.c assumes opcode 0x1b is 32-bit rather than inferring the width from the operands.  I suspect a similar bug may exist for the DROP opcode, though I have not tested that.

wenyongh, 2022-11-16:
Hi, the wasm loader checks whether the operands of opcode SELECT are 32-bit or 64-bit, if they are the latter, loader will modify the original opcode to a newly defined opcode SELECT_64, so aot_compiler.c will handle both SELECT and SELECT_64, and treat the former's operands as 32-bit.

Have you tried running the case with JIT and AOT? I have a test, it works for them.

dbaum, 2022-11-16:
It looks like in my case the opcodes are not being rewritten, but that is
perhaps because I am using both:

WASM_ENABLE_FAST_INTERP=1
and
WASM_ENABLE_JIT=1

It looks like the opcode rewriting is only when WASM_ENABLE_FAST_INTERP==0,
so perhaps that is an unsupported combination.  I'm not familiar with how
this WAMR is being built since another team handled the import and build
integration here.  But if that's unsupported then I'll let them know.

wenyongh, 2022-11-17:
Hi, the fast interpreter will be disabled when llvm jit or fast jit is enabled, since the jit uses the same data structure layout with classic interpter:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/build-scripts/runtime_lib.cmake#L55

dbaum, 2022-11-17:
Thank you for the help!

We can't use cmake and didn't realize WASM_ENABLE_FAST_INTERP and WASM_ENABLE_JIT were mutually exclusive.  Everything works as expected with JIT and the classic interpreter.  Sorry about the misreported bug.

closed 2022-11-17 by dbaum