wasm-micro-runtime/wasm-micro-runtime #1710: Missing invokeNative_em64.asm.obj during MinGW linking

phated, 2022-11-15:
While working on #1680, I encountered an issue when trying to use the assembly code for `invokeNative` on MinGW. When it tries to link the file, it fails with:

```log
Building ASM object CMakeFiles/wamr_static.dir/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.asm.obj
x86_64-w64-mingw32-gcc: warning: wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.asm: linker input file unused because linking not done
```

And then later

```log
/usr/bin/x86_64-w64-mingw32-ar: CMakeFiles/wamr_static.dir/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.asm.obj: No such file or directory
```

I don't have enough experience with C compilers (or cmake) to know what is the problem here. Any thoughts?

wenyongh, 2022-11-25:
@phated I fixed the issue with PR #1753, could you please help have a try?  BTW, I ran several wasi cases, it works, no issues found.

phated, 2022-11-29:
I'm still looking into this. I'm getting unrelated errors from my C compiler now.

phated, 2022-11-29:
@wenyongh did any of your tests use WASI stdin/stdout/stderr? I'm still having trouble with those.

wenyongh, 2022-11-30:
> @wenyongh did any of your tests use WASI stdin/stdout/stderr? I'm still having trouble with those.

Yes, I can compile the asm source code `invokeNative_mingw_x64_simd.s` successfully and build out `iwasm`, and use it to run the wasi case. The source code of the wasi case can be found in attachment `test-wasi.zip`.

[image]
[image]
[test-wasi.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/10118527/test-wasi.zip)

phated, 2022-11-30:
Okay, I need to do more research because I'm getting `"uvwasi init failed"` error when using iwasm core

phated, 2022-12-08:
I've confirmed that this succeeds inside MSYS2 but it doesn't seem to work in Cygwin. I think it is okay to close this, but I still need to figure out how to get this to install inside an OCaml build environment.

closed 2022-12-08 by phated