wasm-micro-runtime/wasm-micro-runtime #1718: WASI directory map not implemented

blackbird1, 2022-11-17:
It seems the "map_dir_list" is not implemented to allow sandboxing the filesystem properly. There is a need to provide virtual mapping to control where wasm apps store their data.

lum1n0us, 2022-11-23:
May I ask at which WASI APIs we are looking? https://github.dev/WebAssembly/WASI/blob/main/phases/ephemeral/witx

blackbird1, 2022-11-23:
`wasm_runtime_set_wasi_args_ex(WASMModuleCommon *module, const char *dir_list[],
                              uint32 dir_count, const char *map_dir_list[],
                              uint32 map_dir_count, const char *env_list[],
                              uint32 env_count, char *argv[], int argc,
                              int stdinfd, int stdoutfd, int stderrfd)` in "wasm_runtime_common.c"

"map_dir_list" is set but never used anywhere in the code to setup the virtual mapping of filesystem paths.

loganek, 2022-11-23:
That looks like that's a miss; I wonder if we can, while fixing this, also expose `--mapdir` parameter in `iwasm` similar to wasmtime so we can also specify mapping through CLI in addition to programmatic interface.

lum1n0us, 2022-11-28:
There will be two steps for implementation:

- If `WAMR_BUILD_LIBC_UVWASI=1`. after taking input argument from `--mapdir`(need to do),  [`mapped_path`]( https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/wasm_runtime_common.c#L2719-L2720) will pass the value to [`uvwasi_preopen_t`](https://github.com/nodejs/uvwasi#uvwasi_preopen_t) (already done). 

- Otherwise, implement mapping filesystem paths in *libc_wasi* (need to do)