wasm-micro-runtime/wasm-micro-runtime #1719: Different Behaviors from native Linux (original) and WASM (WAMR)

chaoyu404, 2022-11-18:
Hello, 
I am trying to merge [llhttp](https://github.com/nodejs/llhttp/tree/release) into my project in the WASM manner, but getting failed. After deep-dive, I realized that [llhttp](https://github.com/nodejs/llhttp/tree/release) was working incorrectly. To clarify the issue, I cloned [llhttp](https://github.com/nodejs/llhttp/tree/release) individually, configurated the testbed and compared [llhttp](https://github.com/nodejs/llhttp/tree/release) behaviors on native Linux and WAMR respectively. 

[llhttp](https://github.com/nodejs/llhttp/tree/release) can parse the test HTTP message successfully on native OS (Linux). However, if we built [llhttp](https://github.com/nodejs/llhttp/tree/release) in the WASM manner and executed it on WAMR, it cannot parse the message and return error, such as "wasm operand stack overflow". It surprised me. Maybe something wrong with WAMR or wasi-sdk? 

To reproduce this issue quickly, I attached the my test package. The test steps are stated in the README.md. Please help to figure out why  [llhttp](https://github.com/nodejs/llhttp/tree/release) got failed. Thanks.
 
[llhttp_test_package.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/10036693/llhttp_test_package.zip)

wenyongh, 2022-11-21:
Hi, "wasm operand stack overflow" means that the wasm operand stack is not enough, could you try enlarging it by `iwasm --stack-size=n`, for example, `iwasm --stack-size=10000000`?

wenyongh, 2022-11-25:
The default wasm operand stack size is enlarged to 64KB by default (#1746), with which we should be able to run the workload without `--stack-size=n` argument.

chaoyu404, 2022-12-01:
Hello 

> The default wasm operand stack size is enlarged to 64KB by default (#1746), with which we should be able to run the workload without `--stack-size=n` argument.

Thanks for your suggestion. 64KB is not enough for llhttp. I enlarged stack to 128KB, and the issue was solved.

closed 2022-12-01 by chaoyu404