wasm-micro-runtime/wasm-micro-runtime #1726: Cannot reproduce Zephyr minimal example

miile7, 2022-11-18:
Hi all,

I tried to reproduce the basic example for Zephyr from the docs (https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#zephyr). I installed Zephyr as described in the (Zephyr) docs. I can build and flash applications, e.g. Zephyr examples to a qemu or a real microcontroller.

The problem is that building WAMR does not work. When I run `./build_and_run.sh qemu` (or `./build_and_run.sh stm32`) I get the following error multiple times after `west build` is being executed (Full output see below):

```
[10/192] Building C object CMakeFiles/app.dir/home/user/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_time.c.obj
FAILED: CMakeFiles/app.dir/home/user/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_time.c.obj 
...
In file included from /home/user/wasm-micro-runtime/core/shared/platform/zephyr/../include/platform_common.h:13,
                 from /home/user/wasm-micro-runtime/core/shared/platform/zephyr/../include/platform_api_vmcore.h:9,
                 from /home/user/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_time.c:6:
/home/user/wasm-micro-runtime/core/shared/platform/zephyr/platform_internal.h:10:10: fatal error: zephyr.h: No such file or directory
   10 | #include <zephyr.h>
      |          ^~~~~~~~~~
compilation terminated.
```

The `$ZEPHYR_BASE` path is set correctly. I executed this in Ubuntu 22.04.1 LTS (in a VM) and reproduced on MacOS Darwin with a fresh installation of Zephyr and WAMR.

Can anyone confirm this behaviour? Or did I do something completely wrong in the installation?

<details>
    <summary>Full Output</summary>

    user@VM:~/wasm-micro-runtime/product-mini/platforms/zephyr/simple (main)$ ./build_and_run.sh stm32
    -- west build: making build dir /home/user/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build pristine
    -- west build: generating a build system
    Loading Zephyr default modules (Zephyr base).
    -- Application: /home/user/wasm-micro-runtime/product-mini/platforms/zephyr/simple
    -- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.6") found components: Interpreter 
    -- Cache files will be written to: /home/user/.cache/zephyr
    -- Zephyr version: 3.2.0-rc2 (/home/user/zephyrproject/zephyr)
    -- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
    -- Board: nucleo_f767zi
    -- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
    -- Found host-tools: zephyr 0.15.0 (/home/user/zephyr-sdk-0.15.0)
    -- Found toolchain: zephyr 0.15.0 (/home/user/zephyr-sdk-0.15.0)
    -- Found Dtc: /home/user/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
    -- Found BOARD.dts: /home/user/zephyrproject/zephyr/boards/arm/nucleo_f767zi/nucleo_f767zi.dts
    -- Generated zephyr.dts: /home/user/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build/zephyr/zephyr.dts
    -- Generated devicetree_generated.h: /home/user/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build/zephyr/include/generated/devicetree_generated.h
    -- Including generated dts.cmake file: /home/user/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build/zephyr/dts.cmake
    Parsing /home/user/zephyrproject/zephyr/Kconfig
    Loaded configuration '/home/user/zephyrproject/zephyr/boards/arm/nucleo_f767zi/nucleo_f767zi_defconfig'
    Configuration saved to '/home/user/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build/zephyr/.config'
    Kconfig header saved to '/home/user/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build/zephyr/include/generated/autoconf.h'
    -- The C compiler identification is GNU 12.1.0
    -- The CXX compiler identification is GNU 12.1.0
    -- The ASM compiler identification is GNU
    -- Found assembler: /home/user/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
    -- Build Configurations:
        Build as target THUMBV7
        CMAKE_BUILD_TYPE 
        WAMR Interpreter enabled
        WAMR AOT enabled
        WAMR Fast JIT disabled
        WAMR LLVM ORC JIT disabled
        Libc builtin enabled
        Libc WASI disabled
        Fast interpreter disabled
        Multiple modules disabled
        Reference types disabled
    -- Configuring done
    -- Generating done
    -- Build files have been written to: /home/user/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build
    -- west build: building application
    [1/192] Preparing syscall dependency handling

[... 109 lines cut ...]
    In file included from /home/user/wasm-micro-runtime/core/shared/utils/../platform/include/platform_common.h:13,
                    from /home/user/wasm-micro-runtime/core/shared/utils/bh_platform.h:9,
                    from /home/user/wasm-micro-runtime/core/shared/mem-alloc/ems/ems_gc_internal.h:13,
                    from /home/user/wasm-micro-runtime/core/shared/mem-alloc/ems/ems_alloc.c:6:
    /home/user/wasm-micro-runtime/core/shared/platform/zephyr/platform_internal.h:10:10: fatal error: zephyr.h: No such file or directory
    10 | #include <zephyr.h>
        |          ^~~~~~~~~~
    compilation terminated.
    [6/183] Building C object CMakeFiles/app.dir/home/user/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_time.c.obj
    FAILED: CMakeFiles/app.dir/home/user/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_time.c.obj 
    ccache /home/user/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBH_FREE=wasm_runtime_free -DBH_MALLOC=wasm_runtime_malloc -DBH_PLATFORM_ZEPHYR -DBUILD_TARGET=\"THUMBV7\" -DBUILD_TARGET_THUMB -DCORE_CM7 -DHSE_VALUE=8000000 -DKERNEL -DSTM32F767xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -DWASM_DISABLE_HW_BOUND_CHECK=0 -DWASM_ENABLE_AOT=1 -DWASM_ENABLE_BULK_MEMORY=0 -DWASM_ENABLE_FAST_INTERP=0 -DWASM_ENABLE_INTERP=1 -DWASM_ENABLE_LIBC_BUILTIN=1 -DWASM_ENABLE_MINI_LOADER=0 -DWASM_ENABLE_MULTI_MODULE=0 -DWASM_ENABLE_SHARED_MEMORY=0 -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/user/wasm-micro-runtime/core/iwasm/interpreter -I/home/user/wasm-micro-runtime/core/iwasm/aot -I/home/user/wasm-micro-runtime/core/iwasm/libraries/libc-builtin -I/home/user/wasm-micro-runtime/product-mini/platforms/zephyr/simple/../../../../core/iwasm/include -I/home/user/wasm-micro-runtime/core/shared/platform/zephyr -I/home/user/wasm-micro-runtime/core/shared/platform/zephyr/../include -I/home/user/wasm-micro-runtime/core/shared/mem-alloc -I/home/user/wasm-micro-runtime/core/iwasm/common -I/home/user/wasm-micro-runtime/core/shared/utils -I/home/user/zephyrproject/zephyr/include -I/home/user/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build/zephyr/include/generated -I/home/user/zephyrproject/zephyr/soc/arm/st_stm32/stm32f7 -I/home/user/zephyrproject/zephyr/drivers -I/home/user/zephyrproject/zephyr/soc/arm/st_stm32/common -I/home/user/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/user/zephyrproject/zephyr/modules/hal_nxp/. -I/home/user/zephyrproject/modules/hal/stm32/stm32cube/stm32f7xx/soc -I/home/user/zephyrproject/modules/hal/stm32/stm32cube/stm32f7xx/drivers/include -I/home/user/zephyrproject/modules/hal/stm32/stm32cube/stm32f7xx/drivers/include/Legacy -I/home/user/zephyrproject/modules/hal/stm32/stm32cube/common_ll/include -isystem /home/user/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/user/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/include -isystem /home/user/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/include-fixed -std=gnu99 -ffunction-sections -fdata-sections                                          -Wall -Wno-unused-parameter -Wno-pedantic -mthumb -fno-strict-aliasing -Os -imacros /home/user/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m7 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/home/user/zephyr-sdk-0.15.0/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/user/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/user/wasm-micro-runtime/product-mini/platforms/zephyr/simple=C [... cut ...]
                    from /home/user/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_time.c:6:
    /home/user/wasm-micro-runtime/core/shared/platform/zephyr/platform_internal.h:10:10: fatal error: zephyr.h: No such file or directory
    10 | #include <zephyr.h>
        |          ^~~~~~~~~~
    compilation terminated.
    ninja: build stopped: subcommand failed.
    FATAL ERROR: re-build in /home/user/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build failed (no --build-dir given)

</details>

Best regards,
miile7

potentialdiffer, 2022-11-18:
Hi all,

I also investigated this issue and I actually found two problems with zephyr build: 

1. The include path for the zephyr header files is incorrect. It must be `<zephyr/zephyr.h>` instead of `<zephyr.h>`. This is true for all zephyr related files. With this corrected, the files compile correctly, however linking fails.
2. Linking fails because of multiple symbol definitions of `sqrt()` and `sqrtf()`. The WASM project provides the "platform math api" which redefines these functions which are already available from zephyr. I got a successful build by removing the two functions named before.

I have no deep insight to the project, but the second issue needs to be integrated into the build setup somehow.

Kind regards
potentialdiffer

goodluckrhj, 2022-11-21:
Hi @miile7 @potentialdiffer 
I think all these errors come from the fact that WAMR's support on Zephyr is dated.
I have meat these problems and asked WAMR for help, but with no problem-solving answers. 
[fatal error: autoconf.h: No such file or directory #1650](https://github.com/bytecodealliance/wasm-micro-runtime/issues/1650)
It's better to seek zephyr official document and espressif solutions to try it out.

Hope the following steps would help, if the latest zephyr v3.2 (v3.2.99 on my ubuntu 22.04 desktop) is to be used:
- Install Zephyr and take WAMR as a zephyr workspace application which enable an easy use of **west** tool (WAMR is tentative one to me, and the Zephyr is always the center role, i will not change the zephyr base or west root).
- The next step is to replace <zephyr.h> by <zephyr/kernel.h>, the process can accelerate if you are using Zephyr's python utility. (zephyr/scripts/utils/[migrate_includes.py](https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/utils/migrate_includes.py) will work.)
- The math re-definition definitely going to be a problem, and function definitions such as sqrt() can be removed to solve the linking error. However, i wonder if it may cause other running problems.

Best regards,
Howard Jen

letanphuc, 2022-11-22:
I make a pull request to fix it here
https://github.com/bytecodealliance/wasm-micro-runtime/pull/1739

miile7, 2022-11-30:
This issue is fixed by #1757

closed 2022-11-30 by miile7