wasm-micro-runtime/wasm-micro-runtime #1727: fast-jit "jit compilation failed: unknown error"

yamt, 2022-11-19:
see https://github.com/yamt/toywasm/blob/424755e4210568ed112b155000c05d772b02e989/test/run-ffmpeg.sh#L26-L33

x86-64 macOS
WAMR commit 021130f4a3377726d9b3ab595caf76214c55fa80
iwasm built with `-DWAMR_BUILD_BULK_MEMORY=1 -DWAMR_BUILD_FAST_JIT=1`

wenyongh, 2022-11-21:
@yamt Thanks for reporting the issue! I uploaded PR #1731 to fix it, could you please try again?

BTW, currently the Fast JIT in main branch uses eager compilation mode, all wasm functions must be compiled before execution, so the startup time may be a little long. You can try the branch `dev/multi_tier_jit`, in which the Fast JIT enables the lazy compilation mode to speedup the launch process. Note that this patch should be applied manually for that branch.

yamt, 2022-11-21:
> @yamt Thanks for reporting the issue! I uploaded PR #1731 to fix it, could you please try again?

i tested it. it worked fine if i gave large enough --jit-codecache-size and --stack-size=20000.

> BTW, currently the Fast JIT in main branch uses eager compilation mode, all wasm functions must be compiled before execution, so the startup time may be a little long. You can try the branch [...]

ok.
i just happened to try fast-jit. i have no real use cases or performance concerns at this point.

closed 2022-11-22 by yamt