wasm-micro-runtime/wasm-micro-runtime #1738: no way to obtain WASI exit code

yamt, 2022-11-22:
a straightforward implementation would introduce:

* wasm_runtime_set_wasi_exit_code
* wasm_runtime_get_wasi_exit_code

and use the former in libc-wasi/uwasi and use the latter in iwasm and other embedders.

if it sounds ok, i can submit an implementation.

wenyongh, 2022-11-23:
@yamt, do you mean to store the wasi exit code in native API `wasi_proc_exit(wasm_exec_env_t exec_env, wasi_exitcode_t rval)` to module instance, and get it from host embedder?

Seems that we only need to provide wasm_runtime_get_wasi_exit_code? Since the wasi exit code should not be set by host embedder.
And could you extend the wasi context to store the exit code? For example, for libc-wasi, add field `wasi_exitcode_t exit_code` in struct WASIContext. For libc-uvwasi, append extra `wasi_exitcode_t exit_code` to the end of uvwasi_t, and malloc uvwas_t with extra buffer when wasm_runtime_init_wasi:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/wasm_runtime_common.c#L2685

yamt, 2022-11-24:
> @yamt, do you mean to store the wasi exit code in native API `wasi_proc_exit(wasm_exec_env_t exec_env, wasi_exitcode_t rval)` to module instance, and get it from host embedder?

yes.

> Seems that we only need to provide wasm_runtime_get_wasi_exit_code? Since the wasi exit code should not be set by host embedder.

right.

> And could you extend the wasi context to store the exit code? For example, for libc-wasi, add field `wasi_exitcode_t exit_code` in struct WASIContext. For libc-uvwasi, append extra `wasi_exitcode_t [...]

i was thinking to store the exit code in module_inst.
but it can be done that way too.

wenyongh, 2022-11-24:
> i was thinking to store the exit code in module_inst. but it can be done that way too.

Hi, to store exit code in module_inst is also OK, but since current AOT module instance layout is strictly defined and should not be changed, there are two ways to store the exit code:
- (1) Store it in the `uint32 reserved[3]` field, e.g.  change to `wasi_exitcode_t exit_code; uint32 reserved[2];`, suppose the size of wasi_exitcode_t is 4 bytes,
- (2) Store it in module_inst->e, the struct of WASMModuleInstanceExtra, but currently the extra info of AOTModuleInstance isn't created.

yamt, 2022-11-24:
fixed

closed 2022-11-24 by yamt