wasm-micro-runtime/wasm-micro-runtime #1743: Are reactors shared between main wasm modules?

alexthomas1, 2022-11-23:
Say I have wasm module A and wasm module B that both have main defined. If I make reactor C, would A and B share reactor C? If not, how do I get them to share?

lum1n0us, 2022-11-25:
If both A and B depend on C, two instances of C will be created. A and B will have their own instance of C. 

A and B can't share reactor C unless A depends on B or else. In this senario, A or B should be another reactor and only one module can be command.

alexthomas1, 2022-11-25:
When I was testing, how come the callback to read an imported module only was called once when I ran module A and B if they are not shared?

lum1n0us, 2022-11-28:
I am afraid I have to take a look at the test code and Wasm modules.