wasm-micro-runtime/wasm-micro-runtime #1755: Using pthread + import doesn't work for some reason..

alexthomas1, 2022-11-25:
```
int add(int x, int y) __attribute__((__import_module__("math"), __import_name__("add")));

void *thread_routine(void *arg)
{
    printf("Enter thread\n");
    pthread_exit(NULL);
    return NULL;
}

int main(int argc, char **argv){
   int z = 0;
   z = add(3, 4);

   pthread_t tid;
   if (0 != pthread_create(&tid, NULL, thread_routine, NULL)) {
        printf("Failed to create thread\n");
    }
   return 0;
}
```

I keep running into seg faults when I use an imported function from another module (reactor) while also trying to use pthread.

lum1n0us, 2022-11-28:
May I ask
- the transform command for *C -> Wasm*
- the compilation combination for *iwasm*
- the running log of the above Wasm applicaiton

alexthomas1, 2022-11-28:
```
/opt/wasi-sdk-16.0/bin/clang++ ${CMAKE_CURRENT_LIST_DIR}/${SOURCE_FILE}.cc
              -O3 -mexec-model=reactor
              -Wl,--max-memory=196608      
              -Wl,--allow-undefined,--no-check-features   
              -Wl,--export=__heap_base,--export=__data_end 
```

This is for the reactor. For the main module it's the same thing except there's not -mexec-model flag and there is a -pthread flag. 

```
/opt/wasi-sdk-16.0/bin/clang++ main.cc
              -O3 -pthread
              -Wl,--max-memory=196608      
              -Wl,--allow-undefined,--no-check-features   
              -Wl,--export=__heap_base,--export=__data_end 
              -o ${WASM_MODULE}
```

The only log I get is 

`Segmentation fault (core dumped)`

lum1n0us, 2022-12-01:
## iwasm with *MULTI_MODULE* and *PTHREAD* supporting

Please make sure to compile *iwasm* with below options ON:

```
set(WAMR_BUILD_INTERP 1)
set(WAMR_BUILD_AOT 1)
set(WAMR_BUILD_JIT 0)
set(WAMR_BUILD_LIBC_BUILTIN 1)
set(WAMR_BUILD_FAST_INTERP 1)
set(WAMR_BUILD_LIB_PTHREAD 1)
set(WAMR_BUILD_LIB_PTHREAD_SEMAPHORE 1)
set(WAMR_BUILD_MULTI_MODULE 1)
```

## Generate .wsm with *WAMR-SDK* instead of *WASI-SDK*

*pthread.h* is not supported in *wasi-sdk-16.0*,

```
set(WAMR_SYS_ROOT ${WAMR_ROOT_DIR}/wamr-sdk/app/libc-builtin-sysroot)

/opt/wasi-sdk-16.0/bin/clang ${CMAKE_CURRENT_LIST_DIR}/reactor_a.c
      --sysroot=${WAMR_SYS_ROOT}
      -nostdlib
      -O3 -mexec-model=reactor
      -Wl,--no-entry
      -Wl,--max-memory=196608
      -Wl,--allow-undefined,--no-check-features
      -Wl,--export=__heap_base,--export=__data_end
      -o reactor.wasm

/opt/wasi-sdk-16.0/bin/clang ${CMAKE_CURRENT_LIST_DIR}/commander_a.c
      --sysroot=${WAMR_SYS_ROOT}
      -nostdlib
      -O3 -pthread
      -Wl,--max-memory=196608
      -Wl,--no-entry
      -Wl,--allow-undefined,--no-check-features
      -Wl,--export=__heap_base,--export=__data_end
      -Wl,--export=__wasm_call_ctors
      -Wl,--export=main -Wl,--export=__main_argc_argv
      -o commander.wasm
```

alexthomas1, 2022-12-01:
Thank you, is there a way to use C++ standard libraries with this?

lum1n0us, 2022-12-02:
like `#include <pthread.h>` -> `#include <thread>` ?

alexthomas1, 2022-12-02:
Like using <string> or <vector>?

lum1n0us, 2022-12-05:
Yes. It is ok to use

- `/opt/wasi-sdk-16.0/bin/clang++` instead of `/opt/wasi-sdk-16.0/bin/clang`
- `/opt/wasi-sdk-16.0/share/wasi-sysroot/` instead of `WAMR_SYS_ROOT`. But need to include *pthread.h* from *wamr-sdk* as an extra help

The new COMMAND will like

```
   /opt/wasi-sdk-16.0/bin/clang++ ${CMAKE_CURRENT_LIST_DIR}/commander.cc
      --sysroot=${WASI_SYS_ROOT}
      -I${WAMR_SYS_ROOT}/include/
  ......
```

alexthomas1, 2022-12-05:
Sorry unsure why the comment was messed up, but I meant to say

LIke using <vector> or <string>?

lum1n0us, 2022-12-06:
Only WASI-libc supported

loganek, 2022-12-06:
@alexthomas1 Under the hood LLVM's libc++ [uses pthread API](https://github.com/llvm/llvm-project/blob/main/libcxx/include/__threading_support) for POSIX platforms, so you should be able to use pthread's WAMR implementation. Having said that, I don't think threads are currently enabled in wasi-sdk, so you might need to have a custom build for that.

alexthomas1, 2022-12-06:
@loganek , thank you. A bit off-topic, but the whole reason I am doing this is to allow modules to dynamically load other modules. Should I save myself the trouble and just proxy dlopen/dlsym to the host? Do reactors work the same way? I still have to pay for the performance penalty of going to the host?

lum1n0us, 2022-12-12:
if enable `WAMR_BUILD_MULTI_MODULE`, WAMR will do a loading-link to search and load all dependencies .wasm if exist. https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/multi_module.md

WASI-libc does support limited range of c++ libs.

closed 2023-04-23 by wenyongh